sdr-enthusiasts/docker-airnavradar #1: Awesome amd64 spin

ShoGinn, 2020-04-04:
Just for clarification they did release the source.. I made attempts to change the source to be multi-arch but it was outside my scope:

https://github.com/mutability/rbfeeder

And the discussion on this source:

https://discussions.flightaware.com/t/exploring-radarbox24s-rpi-feeder/50045/11

mikenye, 2020-04-06:
Hi @ShoGinn, thanks for letting me know about the source. I actually saw this, but given it hadn't been updated since Sept 2019, I wasn't sure it was active.

Also, I don't understand the note at the beginning of the readme:

> To compile this program, you must define the following variables: CONN_KEY, CONN_NONCE and DEF_XOR_KEY.

I'm not sure where to find the value of those variables... :)

I'll remove the bits about not having the source code from the README.md, and for now I think I'll just leave it using qemu until I can figure out the compilation variable.

ShoGinn, 2020-04-06:
What I found with the source is that it’s so embedded in c that it is trying to get serial numbers of the arm processor. Not sure the intent on why they limited it so much. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mikenye, 2020-04-06:
I'm not sure either. Given the qemu method seems to work OK (with the exception of MLAT which I'm working on), being able to compile `rbfeeder` from source is pretty low on my TODO list. :-)

Can I invite you to the adsbexchange discord server? I'm fairly active in the `#docker-help` channel in there and it might make it easier for us to chat about stuff like this. :)

ShoGinn, 2020-04-06:
I’m down. Shoot me an invite 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mikenye, 2020-04-06:
Invite: https://discord.gg/9StvqG

mikenye, 2020-04-06:
Updated the README.md in aformentioned commit.

closed 2020-04-06 by mikenye