sdr-enthusiasts/docker-airnavradar #6: Exec format Error

alexbogus, 2020-05-09:
Hi, 

First of all, thanks for your time in maitaing this cool docker (and others that you have)
I'm trying to deploy the docker to feed to RadarBox, and I'm having the next error

```
,[rbfeeder] ./run: 4: ./run: /usr/bin/rbfeeder: Exec format error
```
My environment

- **Docker version**: 19.03.5
- **Docker-compose version**: 1.20.0-rc2
- **OS**: Ubuntu 64bits into a virtual machine (vmware)

Again, thanks for your time and the effort to maintain the docker
Regards

mikenye, 2020-05-11:
Hi @alexbogus, no problems mate.

I've just now tested the `mikenye/radarbox:latest` image on all supported architectures and it appears to be working ok.

Can you share your `docker run` command or your `docker-compose.yml` file so I can try to find out where the problem lies?

Thanks.

alexbogus, 2020-05-11:
Hi, thanks for help me with this issue
Here is my `docker run`

```
timeout 300s docker run \
    --rm \
    -it \
    -e TZ=Europe\Madrid \
    -e BEASTHOST=readsb \
    -e LAT=xx.xxxx \
    -e LONG=xx.xxx \
    -e ALT=xx \
    --network="default" \
    mikenye/radarbox:latest
```

And this is the output:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-rbfeeder: executing...

WARNING: SHARING_KEY environment variable was not set!
Please make sure you note down the key generated.
Pass the key as environment var SHARING_KEY on next launch!

[cont-init.d] 01-rbfeeder: exited 0.
[cont-init.d] done.
[services.d] starting services
[rbfeeder] ./run: 4: ./run: /usr/bin/rbfeeder: Exec format error
[mlat-client] Delaying mlat-client startup until rbfeeder receives station sn...
[services.d] done.
[rbfeeder] ./run: 4: ./run: /usr/bin/rbfeeder: Exec format error
[rbfeeder] ./run: 4: ./run: /usr/bin/rbfeeder: Exec format error
```

Regards

mikenye, 2020-05-11:
Hi @alexbogus,

No problems.

Can I please ask for the following output from the host running docker:

- `docker image inspect mikenye/radarbox:latest`
- `uname -a`
- `docker run --rm -it --entrypoint file mikenye/radarbox:latest /usr/bin/rbfeeder`
- `docker run --rm -it --entrypoint rbfeeder mikenye/radarbox --version`
- `docker run --rm -it --entrypoint qemu-arm-static mikenye/radarbox -strace /usr/bin/rbfeeder --version`

Thanks.

-Mike

alexbogus, 2020-05-11:
Hi Mike, 
I paste the output of the commands. Here's the info

Output of **docker image inspect mikenye/radarbox:latest**

```
[
    {
        "Id": "sha256:be0b807325f178bc8bf75c88db5c354960fcefed6bf9aa8a28e33e62463f50bc",
        "RepoTags": [
            "mikenye/radarbox:latest"
        ],
        "RepoDigests": [
            "mikenye/radarbox@sha256:1f8a056091e5e085cbae5c0b20f52957a4f7b822907399a6839a64d4e5bbac3b",
            "mikenye/radarbox@sha256:5174ea5af89a24091478cb6873dc03efb22451ae60dd2a561a56dfcb1c05a13a",
            "mikenye/radarbox@sha256:6b44fe0de17d6298b9d74d357fe5e38f94aedc2f9b5b9b7d671b59a9049bbbed",
            "mikenye/radarbox@sha256:98777b7470a38fd3fe20336d384ced7a60decd4ae671db7cec40188a524576cf",
            "mikenye/radarbox@sha256:b25cad584f5143ae81b5b357e4486715c0a4bd5f6aa9c7e0ccdc06a9450d1783"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2020-05-06T15:59:01.880982793Z",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "30105/tcp": {},
                "32088/tcp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "S6_BEHAVIOUR_IF_STAGE2_FAILS=2",
                "BEASTPORT=30005",
                "MLAT_SERVER=mlat1.rb24.com:40900"
            ],
            "Cmd": null,
            "ArgsEscaped": true,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": [
[... 780 lines cut ...]
1 rt_sigprocmask(SIG_SETMASK,0xfffefd64,0xfffefc54) = 0
1 rt_sigaction(SIGILL,0xfffefb0c,0xfffefb98) = 0
1 rt_sigprocmask(SIG_BLOCK,NULL,0xff3a3584) = 0
--- SIGILL {si_signo=SIGILL, si_code=2, si_addr=0xff2bbf68} ---
1 rt_sigprocmask(SIG_SETMASK,0xff3a3584,NULL) = 0
1 rt_sigaction(SIGILL,0xfffefb0c,NULL) = 0
1 rt_sigprocmask(SIG_SETMASK,0xfffefc54,NULL) = 0
1 gettimeofday(-66252,0,1258708034,1258708034,0,-66252) = 0
1 rt_sigaction(SIGPIPE,0xfffefa14,0xfffefaa0) = 0
1 ioctl(1,21523,-66448,1,350088,2) = 0
1 rt_sigaction(SIGINT,0xfffefa14,0xfffefaa0) = 0
1 rt_sigaction(SIGTERM,0xfffefa14,0xfffefaa0) = 0
1 socket(PF_INET,SOCK_DGRAM,IPPROTO_IP) = 3
1 ioctl(3,35111,-67528,234696,3,239096) = 0
1 close(3) = 0
1 gettimeofday(-67660,0,1258708034,1258708034,0,-67660) = 0
1 openat(AT_FDCWD,"/etc/localtime",O_RDONLY|O_CLOEXEC) = -1 errno=2 (No such file or directory)
1 fstat64(1,0xfffef1d8) = 0
1 write(1,0x73f18,39)RBFeeder v0.2.6 (build 20180313082038)
 = 39
1 exit_group(0)
```

Thanks for your support
-Alex

mikenye, 2020-05-11:
OK, I think I've fixed this problem. Can you `docker pull mikenye/radarbox:latest` and try launching your container again? Thanks.

mikenye, 2020-05-11:
Oh one other thing, your `TZ=Europe\Madrid` should be `TZ=Europe/Madrid` (note the forward slash). Cheers.

alexbogus, 2020-05-11:
It's works !!!
Many thanks for your support. what's was the problem?

The output is:

```
[rbfeeder] [2020-05-11 10:30:10]  Starting RBFeeder Version 0.2.6 (build 20180313082038)
[rbfeeder] [2020-05-11 10:30:10]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2020-05-11 10:30:10]  Network-mode enabled.
[rbfeeder] [2020-05-11 10:30:10]                Remote host to fetch data:
[rbfeeder] [2020-05-11 10:30:10]                Remote port: 30005
[rbfeeder] [2020-05-11 10:30:10]                Remote protocol: BEAST
[rbfeeder] [2020-05-11 10:30:11]  System: raspberry
[rbfeeder] [2020-05-11 10:30:11]  Start date/time: 2020-05-11 10:30:10
[rbfeeder] [2020-05-11 10:30:11]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] [2020-05-11 10:30:12]  Connection established.
[rbfeeder] [2020-05-11 10:30:12]  Empty sharing key. We will try to create a new one for you!
```

mikenye, 2020-05-11:
Yay! 

Because RadarBox only supply ARM binaries, I’m using qemu in the container to run them on x86_64. On my development machine, I didn’t need to manually specify the qemu executable (as binfmt detects the arm binary and executed qemu automatically) so I didn’t notice this wasn’t working in the container. I’ve manually put some logic in the service script to manually launch qemu for x86_64. 

Thanks for taking the time to report this as I wouldn’t have found this bug for a long time.

alexbogus, 2020-05-11:
Thankyou to take time to attend me and resolve the issue.

mikenye, 2020-05-11:
You’re very welcome!

closed 2020-05-11 by mikenye