sdr-enthusiasts/docker-airnavradar #9: Segmentation fault (core dumped)

alexbogus, 2020-05-21:
Hi Mike, 

Me again, sorry for the inconvenience.
After a few days of working the docker, I appreciate in the logs a `segmentation fault`.

These are the logs
```
,[mlat-client] Delaying mlat-client startup until rbfeeder receives station sn...
,[rbfeeder] [2020-05-21 16:15:08]  Starting RBFeeder Version 0.2.6 (build 20180313082038)
,[rbfeeder] [2020-05-21 16:15:08]  Using configuration file: /etc/rbfeeder.ini
,[rbfeeder] [2020-05-21 16:15:08]  Network-mode enabled.
,[rbfeeder] [2020-05-21 16:15:08]  		Remote host to fetch data: xx.xx.xx.xx
,[rbfeeder] [2020-05-21 16:15:08]  		Remote port: 30005
,[rbfeeder] [2020-05-21 16:15:08]  		Remote protocol: BEAST
,[rbfeeder] [2020-05-21 16:15:08]  System: raspberry
,[rbfeeder] [2020-05-21 16:15:08]  Start date/time: 2020-05-21 16:15:07
,[rbfeeder] [2020-05-21 16:15:08]  Socket for ANRB created. Waiting for connections on port 32088
,[rbfeeder] [2020-05-21 16:15:09]  Connection established.
,[rbfeeder] [2020-05-21 16:15:10]  Connection with RadarBox24 server OK! Key accepted by server.
,[rbfeeder] [2020-05-21 16:15:38]  ******** Statistics updated every 30 seconds ********
,[rbfeeder] [2020-05-21 16:15:38]  Packets sent in the last 30 seconds: 40, Total packets sent since startup: 40
,[mlat-client] Thu May 21 16:15:38 2020 mlat-client 0.2.11 starting up
,[mlat-client] Thu May 21 16:15:38 2020 Listening for Beast-format results connection on port 30105
,[mlat-client] Thu May 21 16:15:38 2020 Connected to multilateration server at mlat1.rb24.com:40900, handshaking
,Segmentation fault (core dumped)
```
My apologies for the inconvenience
Thanks again four your support

Kind regards
Alex

mikenye, 2020-05-22:
Hi @alexbogus, no inconvenience mate, good to hear from you again.

Firstly, is your underlying host still this?:

```
Linux myhostname 5.3.0-46-generic #38-Ubuntu SMP Fri Mar 27 17:37:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

Secondly, was this a once off or does it happen over and over continually?

Lastly, can you please post the output of `docker inspect rbfeeder` (you might need to change `rbfeeder` to your container name)?

Please santiside this output by censoring your `ALT`, `LAT`, `LONG` and `SHARING_KEY`.

Thanks.

-Mike

alexbogus, 2020-05-22:
Hi Mike, 
Thanks for your support.

Output of my `uname -a`:

`Linux myhostname 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

Output of my rbfeeder container:

```
[
    {
        "Id": "78fd9a5168584cdd8cc475bead93c8318e10306e196d60c7777b0e5ce0a84aad",
        "Created": "2020-05-22T06:08:16.692389629Z",
        "Path": "/init",
        "Args": [],
        "State": {
            "Status": "running",
            "Running": true,
            "Paused": false,
            "Restarting": false,
            "OOMKilled": false,
            "Dead": false,
            "Pid": 2049801,
            "ExitCode": 0,
            "Error": "",
            "StartedAt": "2020-05-22T06:08:17.564829414Z",
            "FinishedAt": "0001-01-01T00:00:00Z"
        },
        "Image": "sha256:d8fffc8c09285f9758e5646cdd6b90efd28730586eaca100598576cc1534b0b3",
        "ResolvConfPath": "/var/lib/docker/containers/78fd9a5168584cdd8cc475bead93c8318e10306e196d60c7777b0e5ce0a84aad/resolv.conf",       
        "HostnamePath": "/var/lib/docker/containers/78fd9a5168584cdd8cc475bead93c8318e10306e196d60c7777b0e5ce0a84aad/hostname",
        "HostsPath": "/var/lib/docker/containers/78fd9a5168584cdd8cc475bead93c8318e10306e196d60c7777b0e5ce0a84aad/hosts",
        "LogPath": "/var/lib/docker/containers/78fd9a5168584cdd8cc475bead93c8318e10306e196d60c7777b0e5ce0a84aad/78fd9a5168584cdd8cc475bead93c8318e10306e196d60c7777b0e5ce0a84aad-json.log",
        "Name": "/radarbox",
        "RestartCount": 0,
        "Driver": "overlay2",
        "Platform": "linux",
        "MountLabel": "",
        "ProcessLabel": "",
        "AppArmorProfile": "docker-default",
        "ExecIDs": null,
        "HostConfig": {
            "Binds": [
                "/home/docker/adsb/radarbox/rbfeeder.ini:/etc/rbfeeder.ini:rw"
            ],
            "ContainerIDFile": "",
            "LogConfig": {
                "Type": "json-file",
                "Config": {
                    "max-file": "5",
                    "max-size": "10m"
                }
            },
            "NetworkMode": "adsb",
            "PortBindings": {},
            "RestartPolicy": {
                "Name": "always",
                "MaximumRetryCount": 0
            },
            "AutoRemove": false,
            "VolumeDriver": "",
            "VolumesFrom": [],
            "CapAdd": null,
            "CapDrop": null,
            "Capabilities": null,
            "Dns": null,
            "DnsOptions": null,
            "DnsSearch": null,
            "ExtraHosts": null,
[... 157 lines cut ...]
                    "IPAMConfig": null,
                    "Links": null,
                    "Aliases": [
                        "78fd9a516858",
                        "rbfeeder"
                    ],
                    "NetworkID": "712df2cbee948ea19eb1bb486f5f8dd34951d8d702f8f034d50119bcc74ccef2",
                    "EndpointID": "84794e62e039991b5503d69b947f50be87b82b54272cdcfd0b1504014366c5c7",
                    "Gateway": "172.18.0.1",
                    "IPAddress": "172.18.0.7",
                    "IPPrefixLen": 16,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "MacAddress": "xx:xx:xx:xx:xx:xx",
                    "DriverOpts": null
                }
            }
        }
    }
]
```
Thank you for taking the time to help me
Kind regards
Alex

mikenye, 2020-05-22:
Is the segfault a once off or does it happen over and over continually?

alexbogus, 2020-05-22:
It's happen over and over continually

mikenye, 2020-05-22:
Are you able to post the output of `tail -100 /var/log/kern.log` ?

You may need to censor some of the information in this file... I'm looking for more information on what is causing the segfault.

Thanks.

alexbogus, 2020-05-22:
This is the output of `tail -100 /var/log/kern.log`

 ```
br-cf6a80473db0: port 15(vethbbc9800) entered disabled state
 eth0: renamed from veth9b9ac73
 IPv6: ADDRCONF(NETDEV_CHANGE): vethbbc9800: link becomes ready
 br-cf6a80473db0: port 15(vethbbc9800) entered blocking state
 br-cf6a80473db0: port 15(vethbbc9800) entered forwarding state
 br-cf6a80473db0: port 15(vethbbc9800) entered disabled state
 veth9b9ac73: renamed from eth0
 br-cf6a80473db0: port 15(vethbbc9800) entered disabled state
 device vethbbc9800 left promiscuous mode
 br-cf6a80473db0: port 15(vethbbc9800) entered disabled state
 br-cf6a80473db0: port 15(veth5f8aef7) entered blocking state
 br-cf6a80473db0: port 15(veth5f8aef7) entered disabled state
 device veth5f8aef7 entered promiscuous mode
 br-cf6a80473db0: port 15(veth5f8aef7) entered blocking state
 br-cf6a80473db0: port 15(veth5f8aef7) entered forwarding state
 br-cf6a80473db0: port 15(veth5f8aef7) entered disabled state
 eth0: renamed from veth854e18f
 IPv6: ADDRCONF(NETDEV_CHANGE): veth5f8aef7: link becomes ready
 br-cf6a80473db0: port 15(veth5f8aef7) entered blocking state
 br-cf6a80473db0: port 15(veth5f8aef7) entered forwarding state
 br-cf6a80473db0: port 15(veth5f8aef7) entered disabled state
 veth854e18f: renamed from eth0
 br-cf6a80473db0: port 15(veth5f8aef7) entered disabled state
 device veth5f8aef7 left promiscuous mode
 br-cf6a80473db0: port 15(veth5f8aef7) entered disabled state
 br-cf6a80473db0: port 15(veth61d9f88) entered blocking state
 br-cf6a80473db0: port 15(veth61d9f88) entered disabled state
 device veth61d9f88 entered promiscuous mode
 br-cf6a80473db0: port 15(veth61d9f88) entered blocking state
 br-cf6a80473db0: port 15(veth61d9f88) entered forwarding state
 br-cf6a80473db0: port 15(veth61d9f88) entered disabled state
 eth0: renamed from vethf24b0ce
 IPv6: ADDRCONF(NETDEV_CHANGE): veth61d9f88: link becomes ready
 br-cf6a80473db0: port 15(veth61d9f88) entered blocking state
 br-cf6a80473db0: port 15(veth61d9f88) entered forwarding state
 br-cf6a80473db0: port 15(veth61d9f88) entered disabled state
 vethf24b0ce: renamed from eth0
 br-cf6a80473db0: port 15(veth61d9f88) entered disabled state
 device veth61d9f88 left promiscuous mode
 br-cf6a80473db0: port 15(veth61d9f88) entered disabled state
 br-cf6a80473db0: port 15(veth8e823d7) entered blocking state
 br-cf6a80473db0: port 15(veth8e823d7) entered disabled state
 device veth8e823d7 entered promiscuous mode
 br-cf6a80473db0: port 15(veth8e823d7) entered blocking state
 br-cf6a80473db0: port 15(veth8e823d7) entered forwarding state
 br-cf6a80473db0: port 15(veth8e823d7) entered disabled state
 eth0: renamed from veth8cf97b9
 IPv6: ADDRCONF(NETDEV_CHANGE): veth8e823d7: link becomes ready
 br-cf6a80473db0: port 15(veth8e823d7) entered blocking state
 br-cf6a80473db0: port 15(veth8e823d7) entered forwarding state
 br-cf6a80473db0: port 15(veth8e823d7) entered disabled state
 veth8cf97b9: renamed from eth0
 br-cf6a80473db0: port 15(veth8e823d7) entered disabled state
 device veth8e823d7 left promiscuous mode
 br-cf6a80473db0: port 15(veth8e823d7) entered disabled state
 br-cf6a80473db0: port 15(veth87c69a7) entered blocking state
 br-cf6a80473db0: port 15(veth87c69a7) entered disabled state
 device veth87c69a7 entered promiscuous mode
 br-cf6a80473db0: port 15(veth87c69a7) entered blocking state
 br-cf6a80473db0: port 15(veth87c69a7) entered forwarding state
 br-cf6a80473db0: port 15(veth87c69a7) entered disabled state
 eth0: renamed from veth9c66b03
 IPv6: ADDRCONF(NETDEV_CHANGE): veth87c69a7: link becomes ready
 br-cf6a80473db0: port 15(veth87c69a7) entered blocking state
 br-cf6a80473db0: port 15(veth87c69a7) entered forwarding state
 veth9c66b03: renamed from eth0
 br-cf6a80473db0: port 15(veth87c69a7) entered disabled state
 br-cf6a80473db0: port 15(veth87c69a7) entered disabled state
 device veth87c69a7 left promiscuous mode
 br-cf6a80473db0: port 15(veth87c69a7) entered disabled state
 br-cf6a80473db0: port 15(vethb37a64a) entered blocking state
 br-cf6a80473db0: port 15(vethb37a64a) entered disabled state
 device vethb37a64a entered promiscuous mode
 br-cf6a80473db0: port 15(vethb37a64a) entered blocking state
 br-cf6a80473db0: port 15(vethb37a64a) entered forwarding state
 br-cf6a80473db0: port 15(vethb37a64a) entered disabled state
 eth0: renamed from veth4552c77
 IPv6: ADDRCONF(NETDEV_CHANGE): vethb37a64a: link becomes ready
 br-cf6a80473db0: port 15(vethb37a64a) entered blocking state
 br-cf6a80473db0: port 15(vethb37a64a) entered forwarding state
 br-cf6a80473db0: port 15(vethb37a64a) entered disabled state
 veth4552c77: renamed from eth0
 br-cf6a80473db0: port 15(vethb37a64a) entered disabled state
 device vethb37a64a left promiscuous mode
 br-cf6a80473db0: port 15(vethb37a64a) entered disabled state
 br-cf6a80473db0: port 15(veth64794f1) entered blocking state
 br-cf6a80473db0: port 15(veth64794f1) entered disabled state
 device veth64794f1 entered promiscuous mode
 br-cf6a80473db0: port 15(veth64794f1) entered blocking state
 br-cf6a80473db0: port 15(veth64794f1) entered forwarding state
 br-cf6a80473db0: port 15(veth64794f1) entered disabled state
 eth0: renamed from veth66ad37d
 IPv6: ADDRCONF(NETDEV_CHANGE): veth64794f1: link becomes ready
 br-cf6a80473db0: port 15(veth64794f1) entered blocking state
 br-cf6a80473db0: port 15(veth64794f1) entered forwarding state
 br-cf6a80473db0: port 15(veth64794f1) entered disabled state
 veth66ad37d: renamed from eth0
 br-cf6a80473db0: port 15(veth64794f1) entered disabled state
 device veth64794f1 left promiscuous mode
 br-cf6a80473db0: port 15(veth64794f1) entered disabled state
```

mikenye, 2020-05-22:
Hmm. I can’t see any helpful information in there. The next time you see a segfault could you capture the last 100 lines of `/var/log/syslog`, `/var/log/kern.log` and `dmesg`?

alexbogus, 2020-05-22:
I'm currently reviewing the logs, but I cannot not see any usefull information
But if I move to /`var/lib/docker/overlay2/<overlayID>/merged/run/s6/services/rbfeeder` I can see some qemu core dumps. Maybe those files could be useful

This is the output:

```
86M     qemu_rbfeeder_20200522-104123_2946.core
86M     qemu_rbfeeder_20200522-104022_2930.core
86M     qemu_rbfeeder_20200522-103922_2914.core
86M     qemu_rbfeeder_20200522-103821_2898.core
86M     qemu_rbfeeder_20200522-103720_2882.core
86M     qemu_rbfeeder_20200522-103620_2866.core
86M     qemu_rbfeeder_20200522-103519_2850.core
86M     qemu_rbfeeder_20200522-103418_2834.core
86M     qemu_rbfeeder_20200522-103317_2818.core
86M     qemu_rbfeeder_20200522-103217_2802.core
```
Maybe those file can be useful to debug?

mikenye, 2020-05-22:
Yes please, I’d like to have a look at them. Any way I can get hold of them? Do you need me to set up a drop box? You probably don’t want to post them publicly as they may contain sensitive info.

alexbogus, 2020-05-22:
I'll send you an email with the details to download the `coredumps`

Regards

mikenye, 2020-05-23:
Core analysis:

```
root@a0efe1f26c08:/# file /tmp/rbfeeder-coredump/qemu_rbfeeder_20200522-104930_3074.core
/tmp/rbfeeder-coredump/qemu_rbfeeder_20200522-104930_3074.core: ELF 32-bit LSB core file ARM, version 1 (SYSV), SVR4-style, from 'rbfeeder', real uid: 0, effective uid: 0, real gid: 0, effective gid: 0
root@a0efe1f26c08:/# gdb /usr/bin/rbfeeder /tmp/rbfeeder-coredump/qemu_rbfeeder_20200522-104930_3074.core
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/rbfeeder...(no debugging symbols found)...done.
[New LWP 3082]
[New LWP 3074]
[New LWP 3077]
[New LWP 3078]
[New LWP 3079]
[New LWP 3080]
[New LWP 3081]
[New LWP 3083]
[New LWP 3084]
[New LWP 3085]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `���'.
#0  __isoc99_fscanf (stream=0x0, format=0x390d4 "%f") at isoc99_fscanf.c:30
30      isoc99_fscanf.c: No such file or directory.
[Current thread is 1 (Thread 0xfbbfc440 (LWP 3082))]
(gdb) bt
#0  __isoc99_fscanf (stream=0x0, format=0x390d4 "%f") at isoc99_fscanf.c:30
#1  0x0002a6b8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) l
25      in isoc99_fscanf.c
```

It looks like `rbfeeder` is segfaulting when trying to open a file?

I can see from your `docker inspect` that you have a bind mount for `/etc/rbfeeder.ini`:

```json
"Mounts": [
            {
                "Type": "bind",
                "Source": "/home/docker/adsb/radarbox/rbfeeder.ini",
                "Destination": "/etc/rbfeeder.ini",
                "Mode": "rw",
                "RW": true,
                "Propagation": "rprivate"
            }
        ],
```

Could you please post the contents of this file (censor your position & sharing key), and also the output of the command `stat /home/docker/adsb/radarbox/rbfeeder.ini`.

Thanks.

mikenye, 2020-05-23:
Could you also try running the container without the bind mount for `/etc/rbfeeder.ini`?

mikenye, 2020-05-23:
Could I also please have the output of `cat /sys/class/thermal/thermal_zone0/temp`? Thanks.

alexbogus, 2020-05-23:
> Could you also try running the container without the bind mount for `/etc/rbfeeder.ini`?

Volume lines commented in `docker-compose`
```
  ##volumes:
      ##- ./radarbox/rbfeeder.ini:/etc/rbfeeder.ini
```
And this is the output, another `Segmentation fault (core dumped)`

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
,[s6-init] ensuring user provided files have correct perms...exited 0.
,[fix-attrs.d] applying ownership & permissions fixes...
,[fix-attrs.d] done.
,[cont-init.d] executing container initialization scripts...
,[cont-init.d] 01-rbfeeder: executing... 
,[cont-init.d] 01-rbfeeder: exited 0.
,[cont-init.d] done.
,[services.d] starting services
,[services.d] done.
,[mlat-client] Delaying mlat-client startup until rbfeeder receives station sn...
,[rbfeeder] [2020-05-23 14:02:44]  Starting RBFeeder Version 0.2.6 (build 20180313082038)
,[rbfeeder] [2020-05-23 14:02:44]  Using configuration file: /etc/rbfeeder.ini
,[rbfeeder] [2020-05-23 14:02:44]  Network-mode enabled.
,[rbfeeder] [2020-05-23 14:02:44]  		Remote host to fetch data: xx.xx.xx.xx
,[rbfeeder] [2020-05-23 14:02:44]  		Remote port: 30005
,[rbfeeder] [2020-05-23 14:02:44]  		Remote protocol: BEAST
,[rbfeeder] [2020-05-23 14:02:44]  System: raspberry
,[rbfeeder] [2020-05-23 14:02:44]  Start date/time: 2020-05-23 14:02:44
,[rbfeeder] [2020-05-23 14:02:44]  Socket for ANRB created. Waiting for connections on port 32088
,[rbfeeder] [2020-05-23 14:02:46]  Connection established.
,[rbfeeder] [2020-05-23 14:02:46]  Connection with RadarBox24 server OK! Key accepted by server.
,[rbfeeder] [2020-05-23 14:03:15]  ******** Statistics updated every 30 seconds ********
,[rbfeeder] [2020-05-23 14:03:15]  Packets sent in the last 30 seconds: 0, Total packets sent since startup: 0
,[mlat-client] Sat May 23 14:03:15 2020 mlat-client 0.2.11 starting up
,[mlat-client] Sat May 23 14:03:16 2020 Listening for Beast-format results connection on port 30105
,[mlat-client] Sat May 23 14:03:16 2020 Connected to multilateration server at mlat1.rb24.com:40900, handshaking
,[mlat-client] Sat May 23 14:03:23 2020 Server says: 
,[mlat-client] 
,[mlat-client]         AirNAv Server
,[mlat-client] 
,[mlat-client]         The multilateration server source code is available under
,[mlat-client]         the terms of the Affero GPL (v3 or later). You may obtain
,[mlat-client]         a copy of this server's source code at the following
,[mlat-client]         location: https://github.com/mutability/mlat-server
,[mlat-client]         
,[mlat-client] Sat May 23 14:03:23 2020 Handshake complete.
,[mlat-client] Sat May 23 14:03:23 2020   Compression:       zlib2
,[mlat-client] Sat May 23 14:03:23 2020   UDP transport:     disabled
,[mlat-client] Sat May 23 14:03:23 2020   Split sync:        disabled
,[mlat-client] Sat May 23 14:03:23 2020 Input connected to xx.xx.xx.xx:30005
,[mlat-client] Sat May 23 14:03:23 2020 Input format changed to BEAST, 12MHz clock
,Segmentation fault (core dumped)
,[rbfeeder] [2020-05-23 14:03:45]  Starting RBFeeder Version 0.2.6 (build 20180313082038)
,[rbfeeder] [2020-05-23 14:03:45]  Using configuration file: /etc/rbfeeder.ini
,[rbfeeder] [2020-05-23 14:03:45]  Network-mode enabled.
,[rbfeeder] [2020-05-23 14:03:45]  		Remote host to fetch data: xx.xx.xx.xx
,[rbfeeder] [2020-05-23 14:03:45]  		Remote port: 30005
,[rbfeeder] [2020-05-23 14:03:45]  		Remote protocol: BEAST
,[rbfeeder] [2020-05-23 14:03:46]  System: raspberry
,[rbfeeder] [2020-05-23 14:03:46]  Start date/time: 2020-05-23 14:03:45
,[rbfeeder] [2020-05-23 14:03:46]  Socket for ANRB created. Waiting for connections on port 32088
,[rbfeeder] [2020-05-23 14:03:47]  Connection established.
```

alexbogus, 2020-05-23:
> Could I also please have the output of `cat /sys/class/thermal/thermal_zone0/temp`? Thanks.

My system It's working dockerized in a VM inside a ESXi Server (v6.5)

mikenye, 2020-05-23:
> > Could I also please have the output of `cat /sys/class/thermal/thermal_zone0/temp`? Thanks.
> 
> My system It's working dockerized in a VM inside a ESXi Server (v6.5)

Could you SSH into the VM and run the command `cat /sys/class/thermal/thermal_zone0/temp`, and post the output?

alexbogus, 2020-05-23:
Here is the output of `stat /home/docker/adsb/radarbox/rbfeeder.ini`

```
Fichero: /home/docker/adsb/radarbox/rbfeeder.ini
Tamaño: 283           Bloques: 8          Bloque E/S: 4096   fichero regular
Dispositivo: 811h/2065d Nodo-i: 26214432    Enlaces: 1
Acceso: (0664/-rw-rw-r--)  Uid: ( 1000/  myusername)   Gid: ( 1000/  myusername)
Acceso: 2020-05-23 16:01:52.470883362 +0200
Modificación: 2020-05-23 16:01:33.882753643 +0200
Cambio: 2020-05-23 16:01:33.882753643 +0200
Creación: -
```

SSh into the VM, and I can't see  ` /sys/class/thermal/thermal_zone0/temp`
I only can see In the /sys/class/thermal directory:

```
drwxr-xr-x  2 root root 0 may 23 16:01 ./
drwxr-xr-x 66 root root 0 may 23 12:23 ../
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device0 -> ../../devices/virtual/thermal/cooling_device0/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device1 -> ../../devices/virtual/thermal/cooling_device1/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device2 -> ../../devices/virtual/thermal/cooling_device2/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device3 -> ../../devices/virtual/thermal/cooling_device3/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device4 -> ../../devices/virtual/thermal/cooling_device4/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device5 -> ../../devices/virtual/thermal/cooling_device5/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device6 -> ../../devices/virtual/thermal/cooling_device6/
lrwxrwxrwx  1 root root 0 may 23 16:01 cooling_device7 -> ../../devices/virtual/thermal/cooling_device7/
```

But doing a recursive search, I cannot find any temp files

mikenye, 2020-05-23:
Hi @alexbogus, I think the problem is with the missing `/sys/class/thermal/thermal_zone0/temp`.

Looking at the rbfeeder source code, searching the files for `fscanf` (shown as the source of the segfault by `gdb` above), we can see the following:

```c
...
    thermal = fopen("/sys/class/thermal/thermal_zone0/temp", "r");
    n = fscanf(thermal, "%f", &millideg);
...
```
Reference: https://github.com/mutability/rbfeeder/blob/8f002217ffe5d5dffced9845bed117ffe2a95a0e/airnav.c#L2529

I think what's happenning is because this code is designed to be run on an RPi (RadarBox don't make binaries for other platforms), I'm using `qemu-arm-static` to run their binaries on an `x86_64` platform. `rbfeeder` is looking for the file `/sys/class/thermal/thermal_zone0/temp` which doesn't exist on your VM (likely because it's virtualised), and causing a segmentation fault.

On my Linux host that I run this on (physical), I have a kernel module loaded called `x86_pkg_temp_thermal`:

```
$ lsmod | grep therm
x86_pkg_temp_thermal    16384  0
```

Could you run the `lsmod | grep therm` command in your VM and see if this module is loaded?

If not, you could try loading it with `modprobe x86_pkg_temp_thermal` and see if that creates the `/sys/class/thermal/thermal_zone0/temp` file.

mikenye, 2020-05-23:
Alternatively, you could try running the commands:

```shell
mkdir -p /sys/class/thermal/thermal_zone0
echo 24000 > /sys/class/thermal/thermal_zone0/temp
```

...which will "spoof" a temperature of 24 degrees C and hopefully fool `rbfeeder`. :)

alexbogus, 2020-05-23:
`mkdir -p /sys/class/thermal/thermal_zone0`
mkdir: cannot create the directory «/sys/class/thermal/thermal_zone0»: Operation not permitted

Ups, I'm trying to pass from host to virtual machine information about temperature.

mikenye, 2020-05-23:
OK try this.

```shell
echo 24000 > /home/docker/adsb/radarbox/faketemp
```

Then add a bind mount to your docker-compose:

under ` - /home/docker/adsb/radarbox/rbfeeder.ini:/etc/rbfeeder.ini`, add:

```yaml
  - /home/docker/adsb/radarbox/faketemp:/sys/class/thermal/thermal_zone0/temp:ro
```

Then recreate the container with `docker-compose up -d`.

alexbogus, 2020-05-24:
Trying the workaround:

`ERROR: for rbfeeder  Cannot start service rbfeeder: OCI runtime create failed: container_linux.go:346: starting container process caused "process_linux.go:449: container init caused \"rootfs_linux.go:58: mounting \\\"/home/docker/adsb/faketemp\\\" to rootfs \\\"/var/lib/docker/overlay2/5d6ed3cf951ef291336be45fc4dbc406f162d07309aa5f75f998c7a390b4a41c/merged\\\" at \\\"/var/lib/docker/overlay2/5d6ed3cf951ef291336be45fc4dbc406f162d07309aa5f75f998c7a390b4a41c/merged/sys/class/thermal/thermal_zone0/temp\\\" caused \\\"mkdir /var/lib/docker/overlay2/5d6ed3cf951ef291336be45fc4dbc406f162d07309aa5f75f998c7a390b4a41c`

mikenye, 2020-05-26:
Can you share your updated `docker-compose.yml` file? The workaround worked in my environment, although I already had the `/sys/class/thermal/thermal_zone0/temp` file present on my system...

Outside of the suggested workarounds, I'm not sure how to solve this problem without either:

- Modifying the `rbfeeder` source code (which I'm not able to do); or
- Somehow get CPU/system temperature into your VM

Sorry Alex. :(

alexbogus, 2020-05-26:
Thanks Mike, I really appreciate your effort to help me with the issue. No problem if we cannot solve this issue.

This is my docker-compose.yml

```
  rbfeeder:
    image: mikenye/radarbox:latest
    logging:
        driver: "json-file"
        options:
            max-file: "5"
            max-size: 10m
    tty: true
    container_name: radarbox
    restart: always
    environment:
      - TZ=${TZ}
      - LAT=${LATITUDE}
      - LONG=${LONGITUDE}
      - BEASTHOST=${READSBHOST}
      - ALT=${ALTITUDE}
      - SHARING_KEY=${RADARBOXKEY}
    volumes:
      - ./radarbox/rbfeeder.ini:/etc/rbfeeder.ini
      - ./radarbox/faketemp:/sys/class/thermal/thermal_zone0/temp:ro
    networks:
      - adsb
```

alexbogus, 2020-05-28:
Consider this issue closed as no further resolution is possible until Radarbox releases new code.

closed 2020-05-28 by alexbogus