sdr-enthusiasts/docker-airnavradar #10: Disk space usage

beercity, 2020-05-26:
Somehow this container consumed 49GB in 9 hours.  

df -h shows:

```
root@8d4f5bb4bad2:/etc# df -h
Filesystem      Size  Used Avail Use% Mounted on
overlay          63G   60G     0 100% /
tmpfs            64M     0   64M   0% /dev
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
/dev/sda2        63G   60G     0 100% /etc/hosts
shm              64M     0   64M   0% /dev/shm
tmpfs           2.0G     0  2.0G   0% /proc/acpi
tmpfs           2.0G     0  2.0G   0% /proc/scsi
tmpfs           2.0G     0  2.0G   0% /sys/firmware
```

I do not quite understand the sda2 mount, I looked at /etc/hosts, and the file is only a few k in size. Quite likely this is a Docker thing that I do not understand.

```
root@8d4f5bb4bad2:/etc# ls -la hosts
-rw-r--r-- 1 root root 177 May 25 14:32 hosts
root@8d4f5bb4bad2:/etc# cat hosts
127.0.0.1       localhost
::1     localhost ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
192.168.160.2   8d4f5bb4bad2
```

I am not sure if this is a container issue or an issue with the code from radarbox

I re-created the container and I will see what looks like in the morning.

beercity, 2020-05-26:
Forgot Docker size output.  This Docker host has 64GB of storage so it hit the limit

`8d4f5bb4bad2        mikenye/radarbox:latest                         "/init"                  9 hours ago         Up 9 hours             30105/tcp, 32088/tcp                                                                                   rbfeeder                     48.8GB (virtual 49.1GB)`

mikenye, 2020-05-26:
Hi @beercity, I don't think this is a problem with the container.

I've had mine running for ~81 days and it's only using ~302MB.

```
# docker exec rbfeeder uptime
 02:53:21 up 81 days, 15:32,  0 users,  load average: 3.52, 4.00, 4.04

# docker exec rbfeeder du -hs /
302M    /
```

Have you set up container log rotation? It is likely that logs for the container are filling up your disk. You can check this with the command:

```
# find /var/lib/docker/containers -type f -name "*.log" -exec ls -lh {} \;
```

To set up container log rotation within Docker, see: https://success.docker.com/article/how-to-setup-log-rotation-post-installation

Further to the link above, I also use a specific logrotate entry for docker logs:

```
# cat /etc/logrotate.d/docker-logs
/var/lib/docker/containers/*/*.log {
  rotate 7
  daily
  compress
  size=1M
  missingok
  delaycompress
  copytruncate
}
```

mikenye, 2020-05-26:
Also, have you checked the container log? I know rbfeeder is pretty noisy normally, however is there something going on causing it to eat up disk space more than usual?

beercity, 2020-05-26:
Thanks for the tip on log rotation.  I did not realize Docker was not doing this automatically, looks like an easy thing to implement.

Last night I looked at the logs and didnt catch it, but it looks like rbfeeder is crashing every minute and being extra noisy about it.  I am not sure how this ate 40GB in 9 hours, but here is a bit of logs.  Note seg fault entries

```

{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  Starting RBFeeder Version 0.2.6 (build 20180313082038)\r\n","stream":"stdout","time":"2020-05-26T11:29:02.638140778Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  Using configuration file: /etc/rbfeeder.ini\r\n","stream":"stdout","time":"2020-05-26T11:29:02.640854314Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  Network-mode enabled.\r\n","stream":"stdout","time":"2020-05-26T11:29:02.641339573Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  \u0009\u0009Remote host to fetch data: 192.168.1.60\r\n","stream":"stdout","time":"2020-05-26T11:29:02.642148974Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  \u0009\u0009Remote port: 30005\r\n","stream":"stdout","time":"2020-05-26T11:29:02.642949347Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  \u0009\u0009Remote protocol: BEAST\r\n","stream":"stdout","time":"2020-05-26T11:29:02.643685896Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  System: raspberry\r\n","stream":"stdout","time":"2020-05-26T11:29:02.768692039Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  Start date/time: 2020-05-26 11:29:02\r\n","stream":"stdout","time":"2020-05-26T11:29:02.769190815Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:02\u001b[35m]\u001b[0m  Socket for ANRB created. Waiting for connections on port 32088\r\n","stream":"stdout","time":"2020-05-26T11:29:02.793332828Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:07\u001b[35m]\u001b[0m  Connection established.\r\n","stream":"stdout","time":"2020-05-26T11:29:07.922167463Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:08\u001b[35m]\u001b[0m  Connection with RadarBox24 server OK! Key accepted by server.\r\n","stream":"stdout","time":"2020-05-26T11:29:08.233772139Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:32\u001b[35m]\u001b[0m  ******** Statistics updated every 30 seconds ********\r\n","stream":"stdout","time":"2020-05-26T11:29:32.797520713Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:29:32\u001b[35m]\u001b[0m  Packets sent in the last 30 seconds: 53, Total packets sent since startup: 53\r\n","stream":"stdout","time":"2020-05-26T11:29:32.798355755Z"}
{"log":"Segmentation fault (core dumped)\r\n","stream":"stdout","time":"2020-05-26T11:30:03.75295273Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:03\u001b[35m]\u001b[0m  Starting RBFeeder Version 0.2.6 (build 20180313082038)\r\n","stream":"stdout","time":"2020-05-26T11:30:03.877895605Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:03\u001b[35m]\u001b[0m  Using configuration file: /etc/rbfeeder.ini\r\n","stream":"stdout","time":"2020-05-26T11:30:03.880568681Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:03\u001b[35m]\u001b[0m  Network-mode enabled.\r\n","stream":"stdout","time":"2020-05-26T11:30:03.881004769Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:03\u001b[35m]\u001b[0m  \u0009\u0009Remote host to fetch data: 192.168.1.60\r\n","stream":"stdout","time":"2020-05-26T11:30:03.881688469Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:03\u001b[35m]\u001b[0m  \u0009\u0009Remote port: 30005\r\n","stream":"stdout","time":"2020-05-26T11:30:03.882436688Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:03\u001b[35m]\u001b[0m  \u0009\u0009Remote protocol: BEAST\r\n","stream":"stdout","time":"2020-05-26T11:30:03.883102168Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:04\u001b[35m]\u001b[0m  System: raspberry\r\n","stream":"stdout","time":"2020-05-26T11:30:04.009173823Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:04\u001b[35m]\u001b[0m  Start date/time: 2020-05-26 11:30:03\r\n","stream":"stdout","time":"2020-05-26T11:30:04.009638494Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:04\u001b[35m]\u001b[0m  Socket for ANRB created. Waiting for connections on port 32088\r\n","stream":"stdout","time":"2020-05-26T11:30:04.033078921Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:09\u001b[35m]\u001b[0m  Connection established.\r\n","stream":"stdout","time":"2020-05-26T11:30:09.165484566Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:09\u001b[35m]\u001b[0m  Connection with RadarBox24 server OK! Key accepted by server.\r\n","stream":"stdout","time":"2020-05-26T11:30:09.373673243Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:34\u001b[35m]\u001b[0m  ******** Statistics updated every 30 seconds ********\r\n","stream":"stdout","time":"2020-05-26T11:30:34.033548668Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:30:34\u001b[35m]\u001b[0m  Packets sent in the last 30 seconds: 71, Total packets sent since startup: 71\r\n","stream":"stdout","time":"2020-05-26T11:30:34.034232324Z"}
[... cut ...]
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:31:35\u001b[35m]\u001b[0m  ******** Statistics updated every 30 seconds ********\r\n","stream":"stdout","time":"2020-05-26T11:31:35.70814993Z"}
{"log":"[rbfeeder] \u001b[35m[\u001b[33m2020-05-26 11:31:35\u001b[35m]\u001b[0m  Packets sent in the last 30 seconds: 64, Total packets sent since startup: 64\r\n","stream":"stdout","time":"2020-05-26T11:31:35.709003276Z"}
{"log":"Segmentation fault (core dumped)\r\n","stream":"stdout","time":"2020-05-26T11:32:06.771317757Z"}
```

beercity, 2020-05-26:
Looks like the seg fault is already covered under issue #9 

The disk space getting eaten up is not logs though.  After ~30 minutes, the container is 3gb, but the log file is only 284182 bytes.  It is the core dumps that are eating the disk space.  88mb each.

Let me know if I can provide any additional help with issue #9

mikenye, 2020-05-26:
Hi @beercity,

I’d suggest taking a look at: https://github.com/mikenye/docker-radarbox/issues/9#issuecomment-633068833

The issue for the segfault is a hard coded entry in the rbfeeder source code to a thermal sensor file, with no error handling if the file doesn’t exit. 

You could try the workarounds I gave Alex in issue #9...

If you have any ideas on how to fix this (without playing with the rbfeeder source code), I’m all ears. :-)

beercity, 2020-05-26:
Yep I am running into the same issue as Alex.  Looks like unless Radarbox will fix the issue, maybe running a job to autodelete the coredump files is a dirty fix?  I am seeing good data in my RB account, so the faults are not totally preventing the container from working.

mikenye, 2020-05-26:
You could try running the ulimit command in the container. 

Try “ulimit -c 0”, which should reduce the number of allowed core dumps to zero. 

If this helps, I’ll add it to the image.

beercity, 2020-05-26:
I added the below to my docker-compose, seems to have stopped the generation of core dumps.
I did ask Radarbox if they would consider releasing code for other platforms, not sure if they take suggestions or not
```
ulimits:
      core: 0
```

alexbogus, 2020-05-27:
Hi, all 

@beercity Hi, I'm having the same issue
@mikenye thanks for the tip to use `ulimit -c 0,` this works for me ok. It's a good workaround since RB solves the issue with the thermal sensor hardcoded.

beercity, 2020-05-27:
Radarbox was receptive to my suggestion to release an x86_64 version.  They gave no commitment on actually doing it, but it might help for anyone with this issue to nicely ask them to release this version.

mikenye, 2020-05-27:
Excellent work!! It would be ideal if we can get x86 and aarch64 versions. Thanks for getting in touch with them.

beercity, 2020-05-28:
Consider this issue closed as no further resolution is possible until Radarbox releases new code.

closed 2020-05-28 by beercity