sdr-enthusiasts/docker-airnavradar #12: Missing /init

echoz, 2020-06-11:
Having issues starting radarbox. I'm pulling the latest from docker hub.

`Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "exec: \"/init\": stat /init: no such file or directory": unknown`

Inspecting the image itself, it does seem that `/init` is missing.

mikenye, 2020-06-11:
Hi @echoz, I've made some code changes to how the architecture is detected during the build process, and this looks to have resolved the problem.

Furthermore, I've added a test in the Dockerfile to make sure an `/init` exists, which will prevent an image from being pushed without `/init` in future.

```
$ docker --context=arm32v7 pull mikenye/radarbox:latest
latest: Pulling from mikenye/radarbox
Digest: sha256:33f24501e954c1fe5c1297fc306cc0e75735ebbd44da64ad47ab73af4784deb3
Status: Downloaded newer image for mikenye/radarbox:latest
docker.io/mikenye/radarbox:latest

$ docker --context=arm32v7 run --rm -it --entrypoint ls mikenye/radarbox:latest -la /init
-rwxr-xr-x 1 root root 389 May 13 18:47 /init

$ docker --context=arm64 pull mikenye/radarbox:latest
latest: Pulling from mikenye/radarbox
Digest: sha256:33f24501e954c1fe5c1297fc306cc0e75735ebbd44da64ad47ab73af4784deb3
Status: Downloaded newer image for mikenye/radarbox:latest
docker.io/mikenye/radarbox:latest

$ docker --context=arm64 run --rm -it --entrypoint ls mikenye/radarbox:latest -la /init
-rwxr-xr-x 1 root root 389 May 13 18:47 /init

$ docker --context=x86_64 pull mikenye/radarbox:latest
latest: Pulling from mikenye/radarbox
Digest: sha256:33f24501e954c1fe5c1297fc306cc0e75735ebbd44da64ad47ab73af4784deb3
Status: Image is up to date for mikenye/radarbox:latest
docker.io/mikenye/radarbox:latest

$ docker --context=x86_64 run --rm -it --entrypoint ls mikenye/radarbox:latest -la /init
-rwxr-xr-x 1 root root 389 May 13 18:47 /init
```

Can you re-pull the latest version of the container, test and let me know if this has solved the problem?

echoz, 2020-06-11:
It works! Thank you for looking into this so quickly!

closed 2020-06-11 by echoz

mikenye, 2020-06-11:
No worries! Thanks for reporting the issue.