sdr-enthusiasts/docker-airnavradar #16: Segmentation fault (core dumped)

avgeeklucky, 2020-06-16:
Not entirely sure how to debug this, but rbfeeder keeps restarting due to a Segmentation fault (see Container Output).

I'm running docker community edition version 19.03.9 on Debian stretch with an AMD64 CPU. Container setup using docker-compose 1.24.1.

<details><summary>docker-compose.yaml</summary>
<p>

```yaml
version: '2.0'

services:
    rbfeeder:
        image: mikenye/radarbox:latest
        restart: always
        container_name: rbfeeder
        restart: always
        environment:
            - TZ=UTC
            - BEASTHOST=192.168.1.20
            - LAT=52.###
            - LONG=4.###
            - ALT=9
            - SHARING_KEY=xxx

```
</p>
</details>
<details><summary>Image Details</summary>
<p>

```json
docker image inspect mikenye/radarbox
[
    {
        "Id": "sha256:375ad043e151c2c0214eec4d31575d678150db2c6df4c68668f4a558da074eb5",
        "RepoTags": [
            "mikenye/radarbox:latest"
        ],
        "RepoDigests": [
            "mikenye/radarbox@sha256:e2e453b50649f3c45df75146eac865df0924c21baba8aa9ff6934283baf5d5f0"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2020-06-15T19:35:10.440867332Z",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
[... 123 lines cut ...]
rbfeeder    | [rbfeeder] [2020-06-16 21:30:55]  ******** Statistics updated every 30 seconds ********
rbfeeder    | [rbfeeder] [2020-06-16 21:30:55]  Packets sent in the last 30 seconds: 0, Total packets sent since startup: 0
rbfeeder    | [rbfeeder] [2020-06-16 21:31:07]  Connection established.
rbfeeder    | [rbfeeder] [2020-06-16 21:31:07]  Connection with RadarBox24 server OK! Key accepted by server.
rbfeeder    | Segmentation fault (core dumped)
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  Starting RBFeeder Version 0.2.6 (build 20180313082038)
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  Using configuration file: /etc/rbfeeder.ini
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  Network-mode enabled.
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  		Remote host to fetch data: 192.168.1.20
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  		Remote port: 30005
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  		Remote protocol: BEAST
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  System: raspberry
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  Start date/time: 2020-06-16 21:31:26
rbfeeder    | [rbfeeder] [2020-06-16 21:31:26]  Socket for ANRB created. Waiting for connections on port 32088
rbfeeder    | [rbfeeder] [2020-06-16 21:31:27]  Connection established.
rbfeeder    | [rbfeeder] [2020-06-16 21:31:37]  Could not start connection. Timeout.
rbfeeder    | [rbfeeder] [2020-06-16 21:31:56]  ******** Statistics updated every 30 seconds ********
rbfeeder    | [rbfeeder] [2020-06-16 21:31:56]  Packets sent in the last 30 seconds: 0, Total packets sent since startup: 0
rbfeeder    | [rbfeeder] [2020-06-16 21:32:07]  Connection established.
rbfeeder    | [rbfeeder] [2020-06-16 21:32:08]  Connection with RadarBox24 server OK! Key accepted by server.
rbfeeder    | Segmentation fault (core dumped)
...
```
</p>
</details>

mikenye, 2020-06-16:
Hi @avgeeklucky,

Have a read of this comment onwards:

https://github.com/mikenye/docker-radarbox/issues/9#issuecomment-633068833

Short version is that the `rbfeeder` code is trying to access a temperature sensor that may not exist on your system (as `rbfeeder` is really only designed to run on a pi) which causes a segfault. 

Unless you can somehow “spoof” this sensor or get it working, without editing the source code for `rbfeeder` there is t much I can do about this unfortunately.

avgeeklucky, 2020-06-16:
Have now. Sorry, completely missed that. Had a look into the spoofing option, /sys does not like to be messed with, that's not going to be easy.

Oh well, I'll just let it crash for now.

closed 2020-06-16 by avgeeklucky

jayp0, 2020-09-27:
mount to /sys/class/thermal instead of /sys/class/thermal/thermal_zone0/temp fixed it.

```
    volumes:
      - ./rb/faketemp:/sys/class/thermal:ro
```
```
$ find rb/faketemp/
rb/faketemp/
rb/faketemp/thermal_zone0
rb/faketemp/thermal_zone0/temp
$ cat rb/faketemp/thermal_zone0/temp
24000
```

mikenye, 2020-09-29:
Fantastic solution, thanks @jayp0