sdr-enthusiasts/docker-airnavradar #24: missing healthcheck

visibilityspots, 2020-07-23:
seems like the healthcheck.sh script is missing in this image;
```
root@3c8709990b8b:~# ls /healthcheck.sh
ls: cannot access '/healthcheck.sh': No such file or directory
```

mikenye, 2020-07-23:
Are you running the latest version of the container?

armv7l:
```
$ docker --context=arm32v7 pull mikenye/radarbox:latest
latest: Pulling from mikenye/radarbox
Digest: sha256:6a7b8a0599e97d52b0620a5f86ad3134a6ca56afb773767da5fa0782c059f12a
Status: Image is up to date for mikenye/radarbox:latest
docker.io/mikenye/radarbox:latest

$ docker --context=arm32v7 run --rm -it --entrypoint ls mikenye/radarbox:latest -la /healthcheck.sh
-rwxr-xr-x 1 root root 1177 Jul 23 09:56 /healthcheck.sh
```

aarch64:
```
$ docker --context=arm64 pull mikenye/radarbox:latest
latest: Pulling from mikenye/radarbox
Digest: sha256:6a7b8a0599e97d52b0620a5f86ad3134a6ca56afb773767da5fa0782c059f12a
Status: Image is up to date for mikenye/radarbox:latest
docker.io/mikenye/radarbox:latest

$ docker --context=arm64 run --rm -it --entrypoint ls mikenye/radarbox:latest -la /healthcheck.sh
-rwxr-xr-x 1 root root 1177 Jul 23 09:56 /healthcheck.sh
```

amd64:
```
$ docker --context=x86_64 pull mikenye/radarbox:latest
latest: Pulling from mikenye/radarbox
Digest: sha256:6a7b8a0599e97d52b0620a5f86ad3134a6ca56afb773767da5fa0782c059f12a
Status: Image is up to date for mikenye/radarbox:latest
docker.io/mikenye/radarbox:latest

$ docker --context=x86_64 run --rm -it --entrypoint ls mikenye/radarbox:latest -la /healthcheck.sh
-rwxr-xr-x 1 root root 1177 Jul 23 09:56 /healthcheck.sh
```

visibilityspots, 2020-07-24:
my bad, had to force pull the latest image and now the file isn't missing anymore. Apologies for the inconvenience!

```
docker run --rm -ti --entrypoint ls mikenye/radarbox:latest -la /healthcheck.sh                                                                                                                  
-rwxr-xr-x 1 root root 1177 Jul 23 09:56 /healthcheck.sh
```

closed 2020-07-24 by visibilityspots

mikenye, 2020-07-24:
No worries at all!