sdr-enthusiasts/docker-airnavradar #25: s6-dnsip4: fatal: unable to resolve

beercity, 2020-07-27:
Having an issue with the latest build, getting error 
`s6-dnsip4: fatal: unable to resolve 192.168.1.60: no such domain`
on container creation

My docker-compose.yml is the same one I was using successfully on the older build
Looks like there is someone on Reddit having the same issue as well.

```
  rbfeeder:
    image: mikenye/radarbox:latest
    tty: true
    container_name: rbfeeder
    restart: always
    environment:
      - TZ=America/Detroit
      - BEASTHOST=192.168.1.1
      - LAT=00.0000
      - LONG=00.0000
      - ALT=000
      - SHARING_KEY=xxxxxxxxxxxxxxx
    networks:
      - adsbnet
    ulimits:
      core: 0
```

mikenye, 2020-07-27:
This should be fixed. Please pull the latest version of the image and let me know if there are any problems. Thanks.

beercity, 2020-07-27:
That did the trick thank you!

closed 2020-07-27 by beercity