sdr-enthusiasts/docker-airnavradar #37: Buffer Overflow Detected

cwagar, 2021-03-17:
vanilla new setup following latest Gitbook HOWTO (March 16, 2021)

rbfeeder is setup with both Radarbox and configured in .env and .yaml according to instructions.

Throws constant Buffer Overflow Detected errors as follows:
```
rbfeeder    | [rbfeeder] *** buffer overflow detected ***: /usr/bin/rbfeeder terminated
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  Starting RBFeeder Version 0.3.5 (build 20200727132301)
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  Using configuration file: /etc/rbfeeder.ini
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  Network-mode enabled.
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  		Remote host to fetch data: 172.18.0.3
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  		Remote port: 30005
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  		Remote protocol: BEAST
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  System: raspberry
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  Start date/time: 2021-03-17 14:21:37
rbfeeder    | [rbfeeder] [2021-03-17 14:21:37]  Socket for ANRB created. Waiting for connections on port 32088
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  Connection established.
rbfeeder    | [rbfeeder] *** buffer overflow detected ***: /usr/bin/rbfeeder terminated
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  Starting RBFeeder Version 0.3.5 (build 20200727132301)
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  Using configuration file: /etc/rbfeeder.ini
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  Network-mode enabled.
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  		Remote host to fetch data: 172.18.0.3
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  		Remote port: 30005
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  		Remote protocol: BEAST
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  System: raspberry
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  Start date/time: 2021-03-17 14:21:38
rbfeeder    | [rbfeeder] [2021-03-17 14:21:38]  Socket for ANRB created. Waiting for connections on port 32088
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  Connection established.
rbfeeder    | [rbfeeder] *** buffer overflow detected ***: /usr/bin/rbfeeder terminated
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  Starting RBFeeder Version 0.3.5 (build 20200727132301)
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  Using configuration file: /etc/rbfeeder.ini
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  Network-mode enabled.
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  		Remote host to fetch data: 172.18.0.3
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  		Remote port: 30005
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  		Remote protocol: BEAST
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  System: raspberry
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  Start date/time: 2021-03-17 14:21:39
rbfeeder    | [rbfeeder] [2021-03-17 14:21:39]  Socket for ANRB created. Waiting for connections on port 32088
```

cwagar, 2021-03-17:
Also, RB24 site is showing my aircraft positions.

mikenye, 2021-03-20:
Hi @cwagar,

I can reproduce this error if I intentionally set my `SHARING_KEY` too long. Can you please ensure your sharing key is set correctly, without any quotes around it?

mikenye, 2021-03-24:
I'm going to close this one out as stale. @cwagar if you still have problems let me know and we can re-open.

closed 2021-03-24 by mikenye

fotomut, 2021-12-31:
For others encountering this issue, I just wanted to share that I had the same problem when setting up my radarbox feeder. I had an extra, invalid character at the end of my `RADARBOX_SHARING_KEY` in my `.env` file.