sdr-enthusiasts/docker-airnavradar #38: Docker container incremental update breaks qemu

ravenium, 2021-03-24:
This can probably be closed, but I figured I'd toss it on here in case someone ran into it.  For some reason about 18 hours ago a new image deploy (ugh, thanks ouroboros) totally hosed my rbfeeder with the below error.  I think it's a golang arch mismatch.  Wiping and repulling the container from scratch this afternoon seemed to fix it, so maybe it was an incremental issue?

standard_init_linux.go:211: exec user process caused "exec format error"

mikenye, 2021-03-25:
Hi @ravenium,

The "exec format error" is usually caused when running a container built for the wrong architecture. I'm wondering if ouroboros did this by mistake?

I'd suggest that you try the architecture-specific release of the container to prevent this from happenning.

ravenium, 2021-03-25:
Ahh, good call - I wonder if it queries arch from the container itself (in which case it would say it's arm) and then since :latest means different things depending on platform...yeah.  

Cheers!

closed 2021-03-25 by ravenium