sdr-enthusiasts/docker-airnavradar #39: Add UAT support

kx1t, 2021-05-25:
Hi Mike, 

RB has started a push for people in the USA to add UAT feeds. They are emailing/giving away UAT dongles and antennas to improve their coverage.
As such, it would be great to add UAT to the capabilities of the container.

It looks like the RadarBox software expects an on-device dump978 instance for this, so you may have to create a SOCAT or similar to connect to your dump978 container.

Here is some info on how to get UAT to work with RadarBox:
https://forum.radarbox24.com/index.php?PHPSESSID=ut24soe2c0eg9ip5i6n0m0c79t&topic=103271.msg444091#msg444091

Note -- the MLAT feeder also isn't working in the container. There's something in the same thread about this. I will do some more research and open a separate issue for this.

Thanks! --R

tedder, 2021-05-25:
interesting, I just dropped them an email in response to that RB email.

mikenye, 2021-06-11:
Hi All,

I have a branch that should now (hopefully) have UAT support. I have no way to test this as I am located in Australia and UAT is not used here.

https://github.com/mikenye/docker-radarbox/tree/add_uat

If you could clone and build the image, then add environment variables `UAT_RECEIVER_HOST` and optionally `UAT_RECEIVER_PORT` (30979), it should hopefully work.

If someone could please test and let me know if it works, I will then PR and merge.

Thanks!

mikenye, 2021-06-11:
# Testing process:

### Clone Repo Locally

```
mkdir -p /home/your_username/docker-radarbox
git clone -b add_uat https://github.com/mikenye/docker-radarbox.git /home/your_username/docker-radarbox
```

### Update Your `docker-compose.yml`

From:

```
  rbfeeder:
    image: mikenye/radarbox:latest
...
```

To:

```
  rbfeeder:
    build: /home/your_username/docker-radarbox
...
```

### Build Image

In the folder that contains your `docker-compose.yml` file (usually `/opt/adsb`), run `docker-compose build`, and wait for the build to finish.

### Run Image

In the folder that contains your `docker-compose.yml` file (usually `/opt/adsb`), run `docker-compose up -d`. Your radarbox container should be recreated and running with the test branch

### Report back

Report back with your findings!

mikenye, 2021-06-11:
If you're testing, once you're running the testing image, please run:

```
docker exec rbfeeder apt-get update
docker exec rbfeeder apt-get install -y net-tools
```

Then, run the following and post the outputs:

```
docker exec rbfeeder cat /etc/rbfeeder.ini | grep -v ^key | grep -v ^lat | grep -v ^lon | grep -v ^alt
docker exec rbfeeder ps aux | grep -v mlat
docker exec rbfeeder netstat -anp
```

Thanks.

mikenye, 2021-06-11:
I’ve had feedback from users that the test branch works, so I’m going to go live with this.

closed 2021-06-14 by mikenye

tedder, 2021-06-14:
[image]

works for me on 'main'.

mikenye, 2021-06-15:
Thanks @tedder!