sdr-enthusiasts/docker-airnavradar #40: Can't generated valid Feeder ID (timeout)

luciodaou, 2021-05-28:
Hi Mike,

I'm having an issue setting up a new receiver based with and amd64 processor (actually an old dell PC with Ubuntu 20.04). I followed the new guide, but I guess there's an error on the first line, should be `radarbox` instead of `piaware`, right?

```
docker pull mikenye/piaware:latest
source ./.env
timeout 60 docker run \
    --rm \
    -it \
    --network adsb_default \
    -e BEASTHOST=readsb \
    -e LAT=${FEEDER_LAT} \
    -e LONG=${FEEDER_LONG} \
    -e ALT=${FEEDER_ALT_M} \
    mikenye/radarbox
```
After running it without any Feeder ID declared on `.env` and no mention to rbfeeder cointainer on `docker-compose.yml`, it gives me an ID and then a lot of timeout errors. The generated ID never are claimable on radarbox website.
[image]

Running without timeout changes nothing:
[image]

When I use an existing Feeder ID with or without the appropriate sections on `docker-compose.yml`, it runs fine without issue.

mikenye, 2021-06-11:
Running the command to generate the sharing key is simply designed to connect to radarbox and generate a sharing key, and that's it.

The command you've shown just runs the feeder for a minute, which is enough time for it to generate the sharing key and then quit.

I can see in your output above, that you have been granted a sharing key. You can ignore all the other errors - that's normal. The only outcome from running that command is to get the sharing key.

luciodaou, 2021-06-15:
Hi Mike,

The problem is that the generated IDs are never validated on RB website. Looks like it needs some real data to be sent to Radarbox, the logs show "0 packets sent".

mikenye, 2021-06-15:
Hi @luciodaou,

Did you go to https://www.radarbox.com/raspberry-pi/claim and claim your feeder?

I just went through the process, and have confirmed that it works:

```
$ timeout 300s docker run     --rm     -it     -e BEASTHOST=x.x.x.x     -e LAT=33.33333     -e LONG=111.11111     -e ALT=30m     mikenye/radarbox
Unable to find image 'mikenye/radarbox:latest' locally
latest: Pulling from mikenye/radarbox
b2cb8f38c044: Already exists
80f3e21bff2a: Pull complete
a722842b2339: Pull complete
Digest: sha256:9c8cdbe64fde034b13f9e0009361ff59c61ccd7d634cff2373e1fd45d3870777
Status: Downloaded newer image for mikenye/radarbox:latest
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-rbfeeder: executing...
WARNING: TZ environment variable not set

WARNING: SHARING_KEY environment variable was not set!
Please make sure you note down the key generated.
Pass the key as environment var SHARING_KEY on next launch!

[cont-init.d] 01-rbfeeder: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[mlat-client] Delaying mlat-client startup until rbfeeder receives station sn...
[rbfeeder] [2021-06-15 02:21:13]  Starting RBFeeder Version 0.4.1 (build 20210525102300)
[rbfeeder] [2021-06-15 02:21:13]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2021-06-15 02:21:13]  Network-mode enabled.
[rbfeeder] [2021-06-15 02:21:13]                Remote host to fetch data: 192.168.69.35
[rbfeeder] [2021-06-15 02:21:13]                Remote port: 30005
[rbfeeder] [2021-06-15 02:21:13]                Remote protocol: BEAST
[rbfeeder] [2021-06-15 02:21:13]  System: raspberry
[rbfeeder] [2021-06-15 02:21:13]  Start date/time: 2021-06-15 02:21:13
[rbfeeder] [2021-06-15 02:21:13]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] [2021-06-15 02:21:15]  Connection established.
[rbfeeder] [2021-06-15 02:21:15]  Empty sharing key. We will try to create a new one for you!
[rbfeeder] [2021-06-15 02:21:16]  Your new key is xxxxxxxxxxx. Please save this key for future use. You will have to know this key to link this receiver to your account in RadarBox24.com. This key is also saved in configuration file (/etc/rbfeeder.ini)
^C[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
s6-svwait: fatal: timed out
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

[image]

closed 2021-06-28 by mikenye