sdr-enthusiasts/docker-airnavradar #43: mlat-client doesn't work with mlathub

tedder, 2021-06-14:
Problem: radarbox claims I've sent zero MLAT messages.

I'm using readsb-protobuf for incoming and a second instance for the mlathub. So I _think_ this doesn't send messages because there is only one BEASTHOST config entry. I mean, [rbfeeder uses BEASTHOST](https://github.com/mikenye/docker-radarbox/blob/0c3fd4f68fc9cae78f0458c20b837a2696b36c71/rootfs/etc/cont-init.d/01-rbfeeder#L14) and [mlat-client also uses BEASTHOST](https://github.com/mikenye/docker-radarbox/blob/d756798cc771c4b99dfba1150d0f3ecf05b99ecb/rootfs/etc/services.d/mlat-client/run#L16).

mikenye, 2021-06-28:
Hi @tedder, there were a couple of PRs related to MLAT over the weekend. Could you pull the latest version and see if that helps?

tedder, 2021-06-28:
I can, but I don't see how those diffs would change anything- there's no way to separate the beast and mlat hosts.

mikenye, 2021-07-04:
Hi @tedder, I might have misunderstood your original issue.

The purpose of mlathub is not to send MLAT messages to feeder containers. It's purpose is to collect the MLAT positions so you can visualise them in your visualtion tool of choice (eg: tar1090).

Radarbox has a lot of MLAT related issues that are no (as far as we can tell) related to the docker container. Reports of no MLAT positions, MLAT connecting and disconnecting repeatedly, etc, come in from many users with different hardware.

closed 2022-02-03 by mikenye