sdr-enthusiasts/docker-airnavradar #49: /usr/bin/rbfeeder: cannot execute binary file: Exec format error

scottwallacesh, 2021-06-27:
v0.4.2 (I think) Docker image isn't working correctly and fails its healthcheck.  Logs show a problem with `rbfeeder` executable format:
```plain
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-rbfeeder: executing... 
[cont-init.d] 01-rbfeeder: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[rbfeeder] 2021/06/27 08:35:37 ./run: line 15: /usr/bin/rbfeeder: cannot execute binary file: Exec format error
[rbfeeder] 2021/06/27 08:35:38 ./run: line 15: /usr/bin/rbfeeder: cannot execute binary file: Exec format error
[rbfeeder] 2021/06/27 08:35:39 ./run: line 15: /usr/bin/rbfeeder: cannot execute binary file: Exec format error
[rbfeeder] 2021/06/27 08:35:40 ./run: line 15: /usr/bin/rbfeeder: cannot execute binary file: Exec format error
[rbfeeder] 2021/06/27 08:35:41 ./run: line 15: /usr/bin/rbfeeder: cannot execute binary file: Exec format error
```

scottwallacesh, 2021-06-27:
I'm running on x86_64.

scottwallacesh, 2021-06-27:
Looks like it's not running `qemu-arm-static` for `x86_64` correctly.

This appears to get it working again:
```diff
--- rootfs/etc/services.d/rbfeeder/run.orig   2021-06-27 08:51:53.722160280 +0100
+++ rootfs/etc/services.d/rbfeeder/run        2021-06-27 08:50:08.944733063 +0100
@@ -12,7 +12,7 @@
         /usr/bin/qemu-arm-static /usr/bin/rbfeeder 2>&1 | \
             stdbuf -oL awk '{print "[rbfeeder] " strftime("%Y/%m/%d %H:%M:%S", systime()) " " $0}'
     else
-        /usr/bin/rbfeeder 2>&1 | \
+        /usr/bin/qemu-arm-static /usr/bin/rbfeeder 2>&1 | \
             stdbuf -oL awk '! / Statistics updated every 30 seconds /' | \
             stdbuf -oL awk '! /Packets sent in the last 30 seconds: /' | \
             stdbuf -oL awk '! / Data sent: /' | \

```

mikenye, 2021-06-27:
Merged, thanks for the PR. Appreciate it.

closed 2021-06-27 by mikenye

scottwallacesh, 2021-06-27:
No problem.

fredclausen, 2021-06-27:
Sorry, I messed that one up.

mikenye, 2021-06-27:
@fredclausen you have nothing to apologise for mate. You got MLAT working. :-)