sdr-enthusiasts/docker-airnavradar #59: qemu: uncaught target signal 11 (Segmentation fault) - core dumped - on new PC

toddy47, 2022-01-26:
Hi Mike,

I recently retired my old server which was running my ADSB system, I had Radarbox running without any issue (had multiple instances for different receivers). Now when I try to build config on the new system, it constantly gives segmentation faults and as a result doesn't load up. 

I tried again on another system and ran into the same problem, I even spun up another docker with the latest pull of docker-radarbox on the original server and it works fine, but the two new servers I constantly get the segmentation faults and no data is ever processed.

Old server was an E3 Xeon, new servers are E5 and E7 Xeons, not sure if that has anything to do with it.

I don't run mikenye/readsb-protobuf, but never did with the old setup either. Tried with a new config and reusing old config with same result.

```

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  System: raspberry

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  Start date/time: 2022-01-26 02:04:33

[rbfeeder] 2022/01/26 02:04:34 [2022-01-26 02:04:34]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:34 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  System: raspberry

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  Start date/time: 2022-01-26 02:04:34

[rbfeeder] 2022/01/26 02:04:35 [2022-01-26 02:04:35]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:35 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  System: raspberry

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  Start date/time: 2022-01-26 02:04:35

[rbfeeder] 2022/01/26 02:04:36 [2022-01-26 02:04:36]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:36 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  System: raspberry

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  Start date/time: 2022-01-26 02:04:36

[rbfeeder] 2022/01/26 02:04:37 [2022-01-26 02:04:37]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:37 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  System: raspberry

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  Start date/time: 2022-01-26 02:04:37

[rbfeeder] 2022/01/26 02:04:38 [2022-01-26 02:04:38]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:38 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  System: raspberry

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  Start date/time: 2022-01-26 02:04:39

[rbfeeder] 2022/01/26 02:04:39 [2022-01-26 02:04:39]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:39 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  System: raspberry

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  Start date/time: 2022-01-26 02:04:40

[rbfeeder] 2022/01/26 02:04:40 [2022-01-26 02:04:40]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/01/26 02:04:40 qemu: uncaught target signal 11 (Segmentation fault) - core dumped

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  Starting RBFeeder Version 0.4.3 (build 20210909231001)

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  Network-mode enabled.

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  		Remote host to fetch data: xx.xx.xx.xx

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  		Remote port: 30005

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  		Remote protocol: RAW

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  System: raspberry

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  Start date/time: 2022-01-26 02:04:41

[rbfeeder] 2022/01/26 02:04:41 [2022-01-26 02:04:41]  Socket for ANRB created. Waiting for connections on port 32088

```

mikenye, 2022-01-26:
Hi @toddy47, have you implemented this segfault fix: https://github.com/mikenye/docker-radarbox/issues/16#issuecomment-699627387 ?

mikenye, 2022-01-26:
More info here: https://mikenye.gitbook.io/ads-b/feeder-containers/feeding-radarbox

toddy47, 2022-01-26:
> More info here: https://mikenye.gitbook.io/ads-b/feeder-containers/feeding-radarbox

Thanks Mike, this indeed solves the issue. I am bit perplexed as to why the original system did not have this fix implemented and had no problem.

Because I have multiple containers for the same websites (IE: 4x receivers) I created a volume with the segment fix and mounted on all of them, seems to be working now.

closed 2022-01-26 by toddy47

mikenye, 2022-01-26:
It might just be differences between CPUs/architectures. The rbfeeder code really should handle the situation better. :-)

Glad it fixed the problem for you.