sdr-enthusiasts/docker-airnavradar #77: Invalid sharing-key

IXEguy, 2022-02-24:
Got readsb running on a pi standalone & fr24, adsbx and fa running in docker yesterday.

Tried to add rbfeeder to the mix and I keep running into this issue. (followed [this](https://mikenye.gitbook.io/ads-b/feeder-containers/feeding-radarbox))

I've tried about 10 times now & waited 5,10,15 minutes after getting the key to  check the claim page & it always says Invalid sharing key in the logs & online. I've check the `rbfeeder.ini` file in the container and it has the correct value. I've copied the value from the output & the ini file and they both come up as invalid.

Nothing in their forums seems to indicate others are having this issue. & I've run out of ideas.

Any suggestions?

Logs:
```
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]  Starting RBFeeder Version 1.0.7 (build 20220223225100)
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]  Network-mode enabled.
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]            Remote host to fetch data: 1.2.3.4
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]            Remote port: 30005
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]            Remote protocol: BEAST
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]  Using GNSS (when available)
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]  Start date/time: 2022-02-24 01:20:13
[rbfeeder] 2022/02/24 01:20:13 [2022-02-24 01:20:13]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/02/24 01:20:14 [2022-02-24 01:20:14]  Connection established.
[rbfeeder] 2022/02/24 01:20:14 [2022-02-24 01:20:14]  Empty sharing key. We will try to create a new one for you!
[rbfeeder] 2022/02/24 01:20:16 [2022-02-24 01:20:16]  Your new key is xxyyzz. Please save this key for future use. You will have to know this key to link this receiver to your account in RadarBox24.com. This key is also saved in configuration file (/etc/rbfeeder.ini)
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 mlat-client 0.2.12 starting up
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 Listening for Beast-format results connection on port 30105
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 Connected to multilateration server at mlat1.rb24.com:40900, handshaking
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 Server says:
[rbfeeder] 2022/02/24 01:20:43
[rbfeeder] 2022/02/24 01:20:43         AirNav Server
[rbfeeder] 2022/02/24 01:20:43
[rbfeeder] 2022/02/24 01:20:43         The multilateration server source code is available under
[rbfeeder] 2022/02/24 01:20:43         the terms of the Affero GPL (v3 or later). You may obtain
[rbfeeder] 2022/02/24 01:20:43         a copy of this server's source code at the following
[rbfeeder] 2022/02/24 01:20:43         location: https://github.com/adsbexchange/mlat-server
[rbfeeder] 2022/02/24 01:20:43
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 Handshake complete.
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022   Compression:       zlib2
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022   UDP transport:     disabled
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022   Split sync:        disabled
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 Input connected to 127.0.0.1:32457
[rbfeeder] 2022/02/24 01:20:43 Thu Feb 24 01:20:43 2022 Input format changed to BEAST, 12MHz clock
[rbfeeder] 2022/02/24 01:20:44 Thu Feb 24 01:20:44 2022 Beast-format results connection with 127.0.0.1:32004: connection established
[rbfeeder] 2022/02/24 01:21:13 [2022-02-24 01:21:13]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/02/24 01:21:13 [2022-02-24 01:21:13]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 1
[rbfeeder] 2022/02/24 01:21:16 [2022-02-24 01:21:16]  Connection established.
[rbfeeder] 2022/02/24 01:21:17 [2022-02-24 01:21:17]  Error authenticating Sharing-Key: Invalid sharing-key
[rbfeeder] 2022/02/24 01:21:27 [2022-02-24 01:21:27]  Could not start connection. Timeout.
```

mikenye, 2022-02-24:
Do you have quotes around your sharing key in your docker-compose.yml file? If so, remove them and see if that helps.

Alternatively, are you able to jump onto our discord server and PM me with your sharing key so I can try to replicate the issue?

Thanks.

mikenye, 2022-02-24:
Also I just noticed the double timestamping in your logs. I'll fix this.

kapiorr, 2022-02-27:
Is problem solved? I have a same issue.

IXEguy, 2022-02-27:
I ended up switching to using radarbox standalone, and that's been working.

However, I am running on an older version on `buster`, not the latest
version on `bullseye`.

Been in touch with Mike. Will try switching over to `1.0.7` standalone (vs
docker) instead of the `0.4.3` later today to see if that works.

mikenye, 2022-03-03:
From what I can gather, the issue is with radarbox. I can't personally replicate the issue, yet it seems to affect some people almost at random. I would suggest reverting to the older version of the container and try again when radarbox release a newer version of rbfeeder... You can revert to the previous version of the container by using tag `mikenye/radarbox:0.4.3-20210909231001`.

jejenone, 2022-03-03:
For what it's worth, I was having the same issue until I end up here. I have used `mikenye/radarbox:0.4.3-20210909231001` to get a new sharing key accepted and it finally worked.
I can use `latest` after my key is accepted.

mikenye, 2022-03-03:
From our Discord discussions regarding this:

The old version (binary release 0.4.3) connects to: `rpi.rb24.com`.
The new version (compiled source code from github repo) connects to `rpiserver-ng.rb24.com`.

Posting here for posterity.

mikenye, 2022-03-03:
For anyone having problems, could you please try removing the CPU serial workaround. I've had some reports in Discord that disabling the CPU workaround for the new `rbfeeder` version seems to correct the problem.

mikenye, 2022-03-03:
I can confirm the feeder works correctly after removing the CPU serial number and temperature sensor workarounds.

kapiorr, 2022-03-03:
How I can remove CPU serial number and temperature sensor from pulled docker image?

mikenye, 2022-03-04:
Remove the `volumes:` section from your container definition in your compose file.

IXEguy, 2022-03-10:
Tried it with just the one volume;
```
  rbfeeder:
    image: mikenye/radarbox:latest
    tty: true
    container_name: rbfeeder
    restart: always
    volumes:
      - "radarbox_segfault_fix:/sys/class/thermal:ro"
    environment:
      - BEASTHOST=${BEASTHOST}
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - ALT=${FEEDER_ALT_M}
      - TZ=${FEEDER_TZ}
      - SHARING_KEY=${RADARBOX_SHARING_KEY}
    tmpfs:
      - /run:exec,size=64M
      - /var/log
```
and no volumes with no luck. Also, pulled latest image from docker just to be sure.

```
[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  Starting RBFeeder Version 1.0.7 (build 20220309123800)

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  Using configuration file: /etc/rbfeeder.ini

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  Network-mode enabled.

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  		Remote host to fetch data: 192.168.86.8

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  		Remote port: 30005

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  		Remote protocol: BEAST

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  Using GNSS (when available)

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  Start date/time: 2022-03-09 21:04:39

[rbfeeder] 2022/03/09 21:04:40 [2022-03-09 21:04:39]  Socket for ANRB created. Waiting for connections on port 32088

[rbfeeder] 2022/03/09 21:04:41 [2022-03-09 21:04:41]  Connection established.

[rbfeeder] 2022/03/09 21:04:51 [2022-03-09 21:04:51]  Could not start connection. Timeout.

[rbfeeder] 2022/03/09 21:05:40 [2022-03-09 21:05:40]  ******** Statistics updated every 60 seconds ********

[rbfeeder] 2022/03/09 21:05:40 [2022-03-09 21:05:40]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 1

[rbfeeder] 2022/03/09 21:05:56 [2022-03-09 21:05:56]  Can't connect to AirNav Server. Retry in 60 seconds.

[rbfeeder] 2022/03/09 21:06:41 [2022-03-09 21:06:41]  ******** Statistics updated every 60 seconds ********

[rbfeeder] 2022/03/09 21:06:41 [2022-03-09 21:06:41]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 1

```

mikenye, 2022-03-25:
See #82 for a workaround.

IXEguy, 2022-03-29:
Finally got the chance & tried the solution from #82 and I am getting the following issue:
```
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  Starting RBFeeder Version 0.4.3 (build 20210909231001)
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  Network-mode enabled.
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  		Remote host to fetch data: 192.168.86.8
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  		Remote port: 30005
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  		Remote protocol: RAW
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  System: raspberry
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  Start date/time: 2022-03-29 12:47:04
[rbfeeder] 2022/03/29 12:47:04 [2022-03-29 12:47:04]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/03/29 12:47:04 qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  Starting RBFeeder Version 0.4.3 (build 20210909231001)
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  Network-mode enabled.
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  		Remote host to fetch data: 192.168.86.8
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  		Remote port: 30005
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  		Remote protocol: RAW
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  System: raspberry
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  Start date/time: 2022-03-29 12:47:05
[rbfeeder] 2022/03/29 12:47:05 [2022-03-29 12:47:05]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/03/29 12:47:05 qemu: uncaught target signal 11 (Segmentation fault) - core dumped
```
Not sure if `qemu: uncaught target signal 11 (Segmentation fault) - core dumped` is related, or a separate issue.

Current docker-compose.yml snippet for this container:
```
  rbfeeder:
    image: mikenye/radarbox:0.4.3-20210909231001
    tty: true
    container_name: rbfeeder
    restart: always
    environment:
      - BEASTHOST=${BEASTHOST}
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - ALT=${FEEDER_ALT_M}
      - TZ=${FEEDER_TZ}
      - SHARING_KEY=${RADARBOX_SHARING_KEY}
    tmpfs:
      - /run:exec,size=64M
      - /var/log
```

I tried pulling the latest radarbox image from the 18th and the issue still persists with that one as well.

mikenye, 2022-03-30:
@IXEguy I believe for the old 0.4.3 version to function correctly, you'll need to implement the temperature sensor workaround outlined here: https://github.com/sdr-enthusiasts/docker-radarbox/tree/26a6bb28398c743f69b7d86479e00f07f1fafec6#workaround-for-temperature-sensor

IXEguy, 2022-04-16:
Sorry, forgot to add that back. 

Finally got the chance to try it out today & it accepted my key that I was using on the Pi.

```
[rbfeeder] 2022/04/16 08:28:03 [2022-04-16 08:28:03]  System: raspberry
[rbfeeder] 2022/04/16 08:28:03 [2022-04-16 08:28:03]  Start date/time: 2022-04-16 08:28:03
[rbfeeder] 2022/04/16 08:28:03 [2022-04-16 08:28:03]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/04/16 08:28:05 [2022-04-16 08:28:05]  Connection established.
[rbfeeder] 2022/04/16 08:28:05 [2022-04-16 08:28:05]  Connection with RadarBox24 server OK! Key accepted by server.
[rbfeeder] 2022/04/16 08:28:05 [2022-04-16 08:28:05]  Your station serial number is: EXTRP1234567
```

So, it works with `mikenye/radarbox:0.4.3-20210909231001` but not with `mikenye/radarbox:latest` or `ghcr.io/sdr-enthusiasts/docker-radarbox:latest`

mikenye, 2022-04-17:
Good news @IXEguy, thanks for your patience with this.

skybarberom, 2022-11-26:
I'm facing the problem described. Is there a newer version of the docker that I can use to get a good key? With the older versions I have a dump

mikenye, 2022-12-15:
@skybarberom can you try with the latest version of the image that was pushed earlier today?

skybarberom, 2022-12-15:
> @skybarberom can you try with the latest version of the image that was pushed earlier today?

@mikenye I made it work last week with the workaround on a docker on QNAP nas.

mikenye, 2022-12-15:
Can this issue be closed?

IXEguy, 2022-12-19:
> Can this issue be closed?

Yes. Thanks for your help on here & 1:1 in resolving some of the issues that I have had!

closed 2022-12-19 by IXEguy

hugalafutro, 2024-01-01:
I think I'm running into this issue. One or two updates of the container back I added the fake cpuinfo and thermal sensor, then it worked for a while, but after updating today I'm getting this invalid key issue. There are no quotation marks around the key in `.env` file.
```
rbfeeder  | s6-rc: info: service s6rc-oneshot-runner: starting
rbfeeder  | s6-rc: info: service s6rc-oneshot-runner successfully started
rbfeeder  | s6-rc: info: service fix-attrs: starting
rbfeeder  | s6-rc: info: service libseccomp2: starting
rbfeeder  | s6-rc: info: service 01-show-rbfeeder-changelog: starting
rbfeeder  | s6-rc: info: service libseccomp2 successfully started
rbfeeder  | s6-rc: info: service fix-attrs successfully started
rbfeeder  | s6-rc: info: service legacy-cont-init: starting
rbfeeder  | s6-rc: info: service 01-show-rbfeeder-changelog successfully started
rbfeeder  | s6-rc: info: service 02-rbfeeder: starting
rbfeeder  | s6-rc: info: service legacy-cont-init successfully started
rbfeeder  | s6-rc: info: service 02-rbfeeder successfully started
rbfeeder  | s6-rc: info: service 03-show-architecture: starting
rbfeeder  | [2024-01-01 17:56:32.857][03-show-architecture] Hardware information:
rbfeeder  | [2024-01-01 17:56:32.857][03-show-architecture] Machine:   x86_64
rbfeeder  | [2024-01-01 17:56:32.857][03-show-architecture] Processor: unknown
rbfeeder  | [2024-01-01 17:56:32.857][03-show-architecture] Platform:  unknown
rbfeeder  | s6-rc: info: service 03-show-architecture successfully started
rbfeeder  | s6-rc: info: service rbfeeder: starting
rbfeeder  | s6-rc: info: service 978json-splitter: starting
rbfeeder  | s6-rc: info: service rbfeeder successfully started
rbfeeder  | s6-rc: info: service stats: starting
rbfeeder  | s6-rc: info: service rbfeeder-log: starting
rbfeeder  | s6-rc: info: service 978json-splitter successfully started
rbfeeder  | s6-rc: info: service stats successfully started
rbfeeder  | s6-rc: info: service rbfeeder-log successfully started
rbfeeder  | s6-rc: info: service legacy-services: starting
rbfeeder  | s6-rc: info: service legacy-services successfully started
rbfeeder  | [2024-01-01 17:56:33.200][rbfeeder] [2024-01-01 17:56:33]  Starting RBFeeder Version 1.0.10 (build 20231120150000)
rbfeeder  | [2024-01-01 17:56:33.201][rbfeeder] [2024-01-01 17:56:33]  Using configuration file: /etc/rbfeeder.ini
rbfeeder  | [2024-01-01 17:56:33.202][rbfeeder] [2024-01-01 17:56:33]  Network-mode enabled.
rbfeeder  | [2024-01-01 17:56:33.202][rbfeeder] [2024-01-01 17:56:33]           Remote host to fetch data: 172.19.0.3
rbfeeder  | [2024-01-01 17:56:33.203][rbfeeder] [2024-01-01 17:56:33]           Remote port: 30005
rbfeeder  | [2024-01-01 17:56:33.203][rbfeeder] [2024-01-01 17:56:33]           Remote protocol: BEAST
rbfeeder  | [2024-01-01 17:56:33.223][rbfeeder] [2024-01-01 17:56:33]  Using GNSS (when available)
rbfeeder  | [2024-01-01 17:56:33.237][rbfeeder] [2024-01-01 17:56:33]  Start date/time: 2024-01-01 17:56:33
rbfeeder  | [2024-01-01 17:56:33.276][rbfeeder] [2024-01-01 17:56:33]  Socket for ANRB created. Waiting for connections on port 32088
rbfeeder  | [2024-01-01 17:56:34.383][rbfeeder] [2024-01-01 17:56:34]  Connection established.
rbfeeder  | [2024-01-01 17:56:34.558][rbfeeder] [2024-01-01 17:56:34]  Error authenticating Sharing-Key: Invalid sharing-key
rbfeeder  | [2024-01-01 17:56:44.516][rbfeeder] [2024-01-01 17:56:44]  Could not start connection. Timeout.
```
relevant part of the ultrafeeer docker-compose:
```
  rbfeeder:
    image: ghcr.io/sdr-enthusiasts/docker-radarbox:latest
    tty: true
    container_name: rbfeeder
    restart: always
    environment:
      - BEASTHOST=ultrafeeder
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - ALT=${FEEDER_ALT_M}
      - TZ=${FEEDER_TZ}
      - SHARING_KEY=${RADARBOX_SHARING_KEY}
    volumes:
      - ./data/fake_cpuinfo:/proc/cpuinfo
      - "radarbox_segfault_fix:/sys/class/thermal:ro"
    tmpfs:
      - /run:exec,size=64M
      - /var/log
    depends_on:
      - ultrafeeder
    labels:
      - "autoheal=true"
```

skybarberom, 2024-01-02:
@hugalafutro I get that message time to time on the first try, but the second try is ok. Did you let it run some minutes to see what it does when it retries?

hugalafutro, 2024-01-02:
@skybarberom that's what usually happens but it was about 2 pages of Invalid sharing key in logs at the time I posted..

But I can see in logs now it connected about half an hour later so I should've been just more patient. Perhaps it was some outage on radarbox side.
```
rbfeeder  | [2024-01-01 18:26:15.082][rbfeeder] [2024-01-01 18:26:15]  Error authenticating Sharing-Key: Invalid sharing-key
rbfeeder  | [2024-01-01 18:26:25.016][rbfeeder] [2024-01-01 18:26:25]  Could not start connection. Timeout.
rbfeeder  | [2024-01-01 18:26:40.845][stats] Packets sent in past 5 minutes: 24, total packets since container start: 25
rbfeeder  | [2024-01-01 18:27:02.697][rbfeeder] [2024-01-01 18:27:02]  ******** Statistics updated every 60 seconds ********
rbfeeder  | [2024-01-01 18:27:02.698][rbfeeder] [2024-01-01 18:27:02]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 26
rbfeeder  | [2024-01-01 18:27:02.698][rbfeeder] [2024-01-01 18:27:02]  Data sent: 1.7 KB
rbfeeder  | [2024-01-01 18:27:02.698][rbfeeder] [2024-01-01 18:27:02]  Data received: 728 B
rbfeeder  | [2024-01-01 18:27:26.259][rbfeeder] [2024-01-01 18:27:26]  Connection established.
rbfeeder  | [2024-01-01 18:27:26.743][rbfeeder] [2024-01-01 18:27:26]  Client type: Raspberry Pi
rbfeeder  | [2024-01-01 18:27:26.761][rbfeeder] [2024-01-01 18:27:26]  Connection with RadarBox24 server OK! Key accepted by server.
```