sdr-enthusiasts/docker-airnavradar #78: dump978 support?

ravenium, 2022-02-24:
Hello,

RB has been sending out a lot of ULAT/978 antennas (basically just rtlsdrs with a nice antenna).  I'd love to try to get docker-radarbox to play nice with the rest of the docker-compose formation and pull 978 info, but am not 100% sure what's involved since RB's documentation is a little opaque and assumes you're running a pi.

I'll dig into it, but has anyone come across a working solution yet?

mikenye, 2022-02-24:
Hi @ravenium,

1. Re-serialise your SDR if required, as per: https://mikenye.gitbook.io/ads-b/setting-up-rtl-sdrs/re-serialise-sdrs
2. Deploy a dump978 container as per: https://mikenye.gitbook.io/ads-b/foundations/deploy-dump978-usa-only
3. Add `      - UAT_RECEIVER_HOST=dump978` to the `environment:` section of your radarbox container. 

That should be all that's required.

Let me know how you go.

ravenium, 2022-02-24:
Thanks @mikenye - looks like I guessed right for most of the steps :)  (and shame on me for not scrolling down on the container's git page)

Looks like I don't need to add   UAT_RECEIVER_TYPE=relay, since that's a flightaware env var?

Other than that, looks like I've got some antenna positioning fun in my future.

mikenye, 2022-02-24:
No you shouldn't need to add that extra variable.

If you run into problems please let me know. The 978 stuff is largely untested by me, as it's not in use where I live.

ravenium, 2022-02-24:
Will do, thanks again.  Radarbox has a nice packet summary per-type, per-day so I'll keep an eye out.

closed 2022-02-24 by ravenium

mikenye, 2022-02-24:
You’re welcome mate. :-)
Good luck.