sdr-enthusiasts/docker-airnavradar #82: Unable to get a sharing key

ON6PH, 2022-03-13:
I am trying to get Radarbox working but seem te get stuck at getting a sharing ID. Here's my output:

root@T630-Ubuntu:/opt/adsb# docker-compose logs -f | grep rbfeeder
Attaching to rbfeeder, tar1090, fr24, piaware, adsbx, readsb
rbfeeder    | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
rbfeeder    | [s6-init] ensuring user provided files have correct perms...exited 0.
rbfeeder    | [fix-attrs.d] applying ownership & permissions fixes...
rbfeeder    | [fix-attrs.d] done.
rbfeeder    | [cont-init.d] executing container initialization scripts...
rbfeeder    | [cont-init.d] 00-libsecomp2: executing...
rbfeeder    | [cont-init.d] 00-libsecomp2: exited 0.
rbfeeder    | [cont-init.d] 01-rbfeeder: executing...
rbfeeder    |
rbfeeder    | WARNING: SHARING_KEY environment variable was not set!
rbfeeder    | Please make sure you note down the key generated.
rbfeeder    | Pass the key as environment var SHARING_KEY on next launch!
rbfeeder    |
rbfeeder    | [cont-init.d] 01-rbfeeder: exited 0.
rbfeeder    | [cont-init.d] done.
rbfeeder    | [services.d] starting services
rbfeeder    | [services.d] done.
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  Starting RBFeeder Version 1.0.7 (build 20220312122700)
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  Using configuration file: /etc/rbfeeder.ini
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  Network-mode enabled.
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  		Remote host to fetch data: 172.24.0.2
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  		Remote port: 30005
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  		Remote protocol: BEAST
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  Using GNSS (when available)
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  Start date/time: 2022-03-13 11:16:09
rbfeeder    | [rbfeeder] 2022/03/13 11:16:09 [2022-03-13 11:16:09]  Socket for ANRB created. Waiting for connections on port 32088
rbfeeder    | [rbfeeder] 2022/03/13 11:16:11 [2022-03-13 11:16:11]  Connection established.
rbfeeder    | [rbfeeder] 2022/03/13 11:16:11 [2022-03-13 11:16:11]  Empty sharing key. We will try to create a new one for you!
rbfeeder    | [rbfeeder] 2022/03/13 11:16:21 [2022-03-13 11:16:21]  Timeout waiting for new key. Will try again in 30 seconds.

And this is the docker-compose.ym part for rbfeeder:
  rbfeeder:
    image: mikenye/radarbox:latest
    tty: true
    container_name: rbfeeder
    restart: always
    depends_on:
      - readsb
    environment:
      - BEASTHOST=readsb
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - ALT=${FEEDER_ALT_M}
      - TZ=${FEEDER_TZ}
      - SHARING_KEY=${RADARBOX_SHARING_KEY}
    tmpfs:
      - /run:exec,size=64M
      - /var/log

I've been talking to Radarbox support but they are not coming up with answers. They've asked me if I use a firewall but I don't. Just a Pi-Hole. Does anyone know how the feeder calls home? Maybe there's a problem with that although I don't see anything relating in my Pi-Hole log.

mikenye, 2022-03-14:
Duplicate of #77

mikenye, 2022-03-25:
Thanks @ON6PH, this is helpful information. I'll leave this issue open in the hopes that new users will find it.