sdr-enthusiasts/docker-airnavradar #96: `Can't connect to AirNav Server.` on Docker build v1.0.7-20220719091600

scottwallacesh, 2022-07-20:
Upgrading to `v1.0.7-20220719091600` results in being unable to connect to the AirNav Server.

I don't know what server it's trying to connect to but reverting the deployed container to `v1.0.7-20220716141600` resolves the problem.

fredclausen, 2022-07-20:
It seems to be an issue on their end. If you give it time the issue appears to resolve itself.

scottwallacesh, 2022-07-20:
Must have just been 'luck' that when I rolled back it was able to connect to the server.  I can now connect (after a couple of container restarts) using the latest image.

It would be helpful to know what address is being connected to so that we can diagnose the problem better on our end.

closed 2022-07-20 by scottwallacesh

fredclausen, 2022-07-20:
I doubt it was 'luck' :) They probably changed something in their binary that was causing problems.

Unfortunately rb_feeder has kind of always been a pain; the actual software used is closed source and good luck getting any support from them. It's kind of always been that way. I took a quick look through the configuration files and there appears to be no configured address anywhere. `netstat` in the container is cutting off the address, but it appears to be some random IP....unfortunately.