sdr-enthusiasts/docker-airnavradar #98: rbfeeder reports that I cannot connect to AirNav Server

echoz, 2022-07-28:
Not sure what I can provide to help better figure this one out, but I guess starting with logs help.

```
Recreating radarbox ... done
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing...
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-rbfeeder: executing...
[cont-init.d] 01-rbfeeder: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[978json-splitter] 2022/07/28 01:05:19 socat[263] W ioctl(5, IOCTL_VM_SOCKETS_GET_LOCAL_CID, ...): Inappropriate ioctl for device
[978json-splitter] 2022/07/28 01:05:19 socat[263] N listening on AF=2 0.0.0.0:30979
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  Starting RBFeeder Version 1.0.7 (build 20220719091400)
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  Network-mode enabled.
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  		Remote host to fetch data: 192.168.1.193
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  		Remote port: 30005
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  		Remote protocol: BEAST
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  Using GNSS (when available)
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  Start date/time: 2022-07-28 01:05:19
[rbfeeder] 2022/07/28 01:05:19 [2022-07-28 01:05:19]  Socket for ANRB created. Waiting for connections on port 32088
[978json-splitter] 2022/07/28 01:05:22 socat[263] N accepting connection from AF=2 127.0.0.1:58596 on AF=2 127.0.0.1:30979
[978json-splitter] 2022/07/28 01:05:22 socat[263] N forked off child process 291
[978json-splitter] 2022/07/28 01:05:22 socat[263] N listening on AF=2 0.0.0.0:30979
[978json-splitter] 2022/07/28 01:05:22 socat[291] N opening connection to AF=2 192.168.1.193:30979
[978json-splitter] 2022/07/28 01:05:22 socat[291] N successfully connected from local address AF=2 172.18.0.3:34676
[978json-splitter] 2022/07/28 01:05:22 socat[291] N starting data transfer loop with FDs [6,6] and [5,5]
[rbfeeder] 2022/07/28 01:05:25 [2022-07-28 01:05:25]  Can't connect to AirNav Server. Retry in 60 seconds.
[rbfeeder] 2022/07/28 01:06:19 [2022-07-28 01:06:19]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/07/28 01:06:19 [2022-07-28 01:06:19]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 0
[rbfeeder] 2022/07/28 01:06:19 [2022-07-28 01:06:19]  Data sent: 0 B
[rbfeeder] 2022/07/28 01:06:19 [2022-07-28 01:06:19]  Data received: 0 B
[rbfeeder] 2022/07/28 01:06:25 [2022-07-28 01:06:25]  Connection established.
```

echoz, 2022-07-28:
I can netcat from within the container to various sites, including `rpi.rb24.com` but not sure why `rbfeeder` just can't connect

mikenye, 2022-12-15:
This should be resolved by the latest version of the image that was built earlier today. Could you please pull the latest version of the image and see if this fixes your problem. Thanks.

closed 2022-12-15 by echoz

mikenye, 2022-12-15:
You’re very welcome @echoz :-)