sdr-enthusiasts/docker-airnavradar #99: Could not start connection. Timeout.

doctaste, 2022-08-06:
I'm having issues connecting to radarbox since a while.

[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]  Starting RBFeeder Version 1.0.7 (build 20220719091400)
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]  Network-mode enabled.
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]            Remote host to fetch data: 172.17.0.1
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]            Remote port: 30005
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]            Remote protocol: BEAST
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]  Using GNSS (when available)
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]  Start date/time: 2022-08-06 08:57:20
[rbfeeder] 2022/08/06 08:57:20 [2022-08-06 08:57:20]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/08/06 08:57:21 [2022-08-06 08:57:21]  Connection established.
[rbfeeder] 2022/08/06 08:57:31 [2022-08-06 08:57:31]  Could not start connection. Timeout.
[rbfeeder] 2022/08/06 08:58:20 [2022-08-06 08:58:20]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/08/06 08:58:20 [2022-08-06 08:58:20]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 1
[rbfeeder] 2022/08/06 08:58:20 [2022-08-06 08:58:20]  Data sent: 63 B
[rbfeeder] 2022/08/06 08:58:20 [2022-08-06 08:58:20]  Data received: 0 B
[rbfeeder] 2022/08/06 08:58:31 [2022-08-06 08:58:31]  Connection established.
[rbfeeder] 2022/08/06 08:58:41 [2022-08-06 08:58:41]  Could not start connection. Timeout.
[rbfeeder] 2022/08/06 08:59:21 [2022-08-06 08:59:21]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/08/06 08:59:21 [2022-08-06 08:59:21]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 2
[rbfeeder] 2022/08/06 08:59:21 [2022-08-06 08:59:21]  Data sent: 126 B
[rbfeeder] 2022/08/06 08:59:21 [2022-08-06 08:59:21]  Data received: 0 B
[rbfeeder] 2022/08/06 08:59:47 [2022-08-06 08:59:47]  Can't connect to AirNav Server. Retry in 60 seconds.
[rbfeeder] 2022/08/06 09:00:22 [2022-08-06 09:00:22]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/08/06 09:00:22 [2022-08-06 09:00:22]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 2
[rbfeeder] 2022/08/06 09:00:22 [2022-08-06 09:00:22]  Data sent: 126 B
[rbfeeder] 2022/08/06 09:00:22 [2022-08-06 09:00:22]  Data received: 0 B
[rbfeeder] 2022/08/06 09:00:53 [2022-08-06 09:00:53]  Can't connect to AirNav Server. Retry in 60 seconds.
[rbfeeder] 2022/08/06 09:01:23 [2022-08-06 09:01:23]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/08/06 09:01:23 [2022-08-06 09:01:23]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 2
[rbfeeder] 2022/08/06 09:01:23 [2022-08-06 09:01:23]  Data sent: 126 B
[rbfeeder] 2022/08/06 09:01:23 [2022-08-06 09:01:23]  Data received: 0 B
[rbfeeder] 2022/08/06 09:01:54 [2022-08-06 09:01:54]  Connection established.
[rbfeeder] 2022/08/06 09:02:04 [2022-08-06 09:02:04]  Could not start connection. Timeout.
[rbfeeder-stats] 2022/08/06 09:02:20 Packets sent in past 5 minutes: 2, total packets since container start: 2
[rbfeeder] 2022/08/06 09:02:24 [2022-08-06 09:02:24]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/08/06 09:02:24 [2022-08-06 09:02:24]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 3
[rbfeeder] 2022/08/06 09:02:24 [2022-08-06 09:02:24]  Data sent: 189 B
[rbfeeder] 2022/08/06 09:02:24 [2022-08-06 09:02:24]  Data received: 0 B
[rbfeeder] 2022/08/06 09:03:10 [2022-08-06 09:03:10]  Can't connect to AirNav Server. Retry in 60 seconds.

It worked fine for a while, but then it stopped to work.I'm under the impression that Airbav does some IP blocking for whatever reason.When i change my public IP to a new one ( i'm using VPN), then it is working again, but only for a while ( can be 4 days, but also sometimes only a few hours).Once this happened with a certain public IP, it will not work anymore with that particular public IP.So i cannot make a successful connection with my ISP public IP anymore.But if i choose a VPN public IP i never used before with radarbox, it will work again.For a limited time, then goes offline again.

I did a network capture and i see two different issues here.

1. _Can't connect to AirNav Server. Retry in 60 seconds._

In this case rpiserver-ng.rb24.com does not respond at all.This looks like a pure Airnav issue, they are using 4 different IP addresses for that hostname and maybe the IP address returned by the DNS lookup is overloaded or some internal network issues etc.

Name:    rpiserver-ng.rb24.com
Addresses:  23.29.123.254
          23.29.123.240
          79.133.49.90
          79.133.49.85

So in my example here 23.29.123.254 did not respond at all, whereas 23.29.123.240 did. 

This is only a temporary failure, which usually resolves on its own on next connection attempts.

2. _Could not start connection. Timeout._

In this case the TCP connection to rpiserver-ng.rb24.com is successful made and the docker client sends a small data packet, which only contains two bytes ( 7e23) and then rpiserver-ng.rb24.com is closing the connection immediately ( FIN+RST).
When it's working, rpiserver-ng.rb24.com responds with ACK to the 7e23 and then the client is sending the real data.
What is 7e23 ? Is this an identifier ?

*******************************************************
Internet Protocol Version 4, Src: Flightradar (192.XXX.XXX.XXX), Dst: rpiserver-ng.rb24.com (23.29.123.240)
Transmission Control Protocol, Src Port: 55528, Dst Port: 33755, Seq: 1, Ack: 1, Len: 2
    Source Port: 55528
    Destination Port: 33755
    [Stream index: 13]
    [Conversation completeness: Incomplete, DATA (15)]
    [TCP Segment Len: 2]
    Sequence Number: 1    (relative sequence number)
    Sequence Number (raw): 1368053075
    [Next Sequence Number: 3    (relative sequence number)]
    Acknowledgment Number: 1    (relative ack number)
    Acknowledgment number (raw): 442404819
    1000 .... = Header Length: 32 bytes (8)
    Flags: 0x018 (PSH, ACK)
    Window: 502
    [Calculated window size: 64256]
    [Window size scaling factor: 128]
    Checksum: 0xaa54 [unverified]
    [Checksum Status: Unverified]
    Urgent Pointer: 0
    Options: (12 bytes), No-Operation (NOP), No-Operation (NOP), Timestamps
    [Timestamps]
    [SEQ/ACK analysis]
    TCP payload (2 bytes)
Data (2 bytes)
    Data: 7e23
    [Length: 2]
*****************************************************************

First i thought about Airnav blocking VPN clients, but i cannot make it work with my normal ISP IP neither anymore.
Could it be that Airnav is blocking the docker client ?

I also checked Radarbox forums / Telegram group, but i couldn't find anything like this.So i assume that the raspberry PI clients don't have that issue.

magnum61, 2022-08-18:
Just installed the docker image on a fresh system for my existing RB Feed (using an existing sharing key).

But i do have the same issue while it is working on a raspberry native. 

I am using a Jetvision Radarcape device as source and added the correct IP/Port.
The Radarcape detected the connect to it's port correctly, so this is not the root cause.

Any help would be appreciated.

yitzyb, 2022-10-27:
just heard from airnav support that the latest version does not support containers.  waiting for an update from them if there is a solution

yitzyb, 2022-11-22:
> just heard from airnav support that the latest version does not support containers. waiting for an update from them if there is a solution

This is now resolved as of yesterday (21-Nov).  As far as I know there were no changes to the container.

magnum61, 2022-11-22:
Nope, not or me. Same problem

```
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]  Starting RBFeeder Version 1.0.7 (build 20221027145200)
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]  Network-mode enabled.
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]            Remote host to fetch data: 192.168.xxx.xxx
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]            Remote port: 10006
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]            Remote protocol: BEAST
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]  Using GNSS (when available)
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]  Start date/time: 2022-11-22 20:08:21
[rbfeeder] 2022/11/22 20:08:21 [2022-11-22 20:08:21]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/11/22 20:08:22 [2022-11-22 20:08:22]  Connection established.
[rbfeeder] 2022/11/22 20:08:32 [2022-11-22 20:08:32]  Could not start connection. Timeout.
[rbfeeder] 2022/11/22 20:09:21 [2022-11-22 20:09:21]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/11/22 20:09:21 [2022-11-22 20:09:21]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 1
[rbfeeder] 2022/11/22 20:09:32 [2022-11-22 20:09:32]  Connection established.
[rbfeeder] 2022/11/22 20:09:43 [2022-11-22 20:09:43]  Could not start connection. Timeout.
```

beercity, 2022-12-01:
i was having this problem with the mikenye repo, i switched to the sdr-enthusiasts repo and it worked right away. check your docker-compose

yitzyb, 2022-12-01:
> i was having this problem with the mikenye repo, i switched to the sdr-enthusiasts repo and it worked right away. check your docker-compose

I am only using the sdr-enthusiasts repos.  actually yesterday radarbox stopped sending data again - same error - could not start connect, timeout.

doctaste, 2022-12-02:
I'm also only using sdr-enthusiasts ( thats why i posted this here ;-) and i'm still having issues with it.

In the oveview in the left hand bottom section of the web ui i get yellow "no data" blocks in the early morning hours every day.Sometimes also red blocks, but it recovers on it's own.As stated in my post above, i see a relationship to my public internet IP address.When using my normal ISP IP, it cannot make a connection. Using a VPN service it can establish a connection, but it depends on the location / IP address of the VPN node / server.With some it's working, with others it's not.

Sometimes i'm seeing something like this:

rbfeeder] 2022/12/02 13:40:48 [2022-12-02 13:40:48]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/12/02 13:40:48 [2022-12-02 13:40:48]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 11
[rbfeeder] 2022/12/02 13:40:48 [2022-12-02 13:40:48]  Data sent: 693 B
[rbfeeder] 2022/12/02 13:40:48 [2022-12-02 13:40:48]  Data received: 0 B
[rbfeeder] 2022/12/02 13:41:41 [2022-12-02 13:41:41]  Connection established.
[rbfeeder] 2022/12/02 13:41:41 [2022-12-02 13:41:41]  Error authenticating Sharing-Key: Invalid sharing-key
[rbfeeder] 2022/12/02 13:41:49 [2022-12-02 13:41:49]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/12/02 13:41:49 [2022-12-02 13:41:49]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 12
[rbfeeder] 2022/12/02 13:41:49 [2022-12-02 13:41:49]  Data sent: 756 B
[rbfeeder] 2022/12/02 13:41:49 [2022-12-02 13:41:49]  Data received: 28 B
[rbfeeder] 2022/12/02 13:41:51 [2022-12-02 13:41:51]  Could not start connection. Timeout.
[rbfeeder] 2022/12/02 13:42:50 [2022-12-02 13:42:50]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/12/02 13:42:50 [2022-12-02 13:42:50]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 12
[rbfeeder] 2022/12/02 13:42:50 [2022-12-02 13:42:50]  Data sent: 756 B
[rbfeeder] 2022/12/02 13:42:50 [2022-12-02 13:42:50]  Data received: 28 B
[rbfeeder] 2022/12/02 13:42:52 [2022-12-02 13:42:52]  Connection established.
[rbfeeder] 2022/12/02 13:42:52 [2022-12-02 13:42:52]  Client type: Raspberry Pi
[rbfeeder] 2022/12/02 13:42:52 [2022-12-02 13:42:52]  Connection with RadarBox24 server OK! Key accepted by server.
[rbfeeder] 2022/12/02 13:42:52 [2022-12-02 13:42:52]  This is your station serial number: EXTRPIXXXXXX

The IP did not change during this.The sharing did neither.

I then setup a real PI client and the behavior is the same.When the PI can connect, the docker radarbox can connect too and vice versa.

nslookup and traceroute returns the same, no matter what public IP i'm using, so there are no DNS or routing issues.

So i still suspect that Airnav servers are blocking some IP ranges.I don't have these issues with any of my other flightrackers ( Flightradar24, Flightaware, Planefinder, adsbexchange)

yitzyb, 2022-12-02:
I also suspect IP and/or geo blocking. I'm using the same box running containers feeding other services with no issue. It randomly started working last week for a few days then stopped again.  I'll test connecting via different ISPs and report back.

mikenye, 2022-12-15:
This should be resolved by the latest version of the image that was built earlier today. Could you please pull the latest version of the image and see if this fixes your problem. Thanks.

yitzyb, 2022-12-15:
@mikenye 
I initially had the cpu serial number issue and generated the random serial and updating docker compose.
after pulling the latest image, it's been over an hour since starting and still not connecting

mikenye, 2022-12-15:
@yitzyb can you post your container log (with any sensitive info removed)?

doctaste, 2022-12-15:
> This should be resolved by the latest version of the image that was built earlier today. Could you please pull the latest version of the image and see if this fixes your problem. Thanks.

For me it doesn't work at all, It seems to pull some armhf image, no matter if i specify latest or amd64 / i386 images.

[rbfeeder] [2022-12-15 15:03:17]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-15 15:03:17]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-15 15:03:17]  Network-mode enabled.
[rbfeeder] [2022-12-15 15:03:17]                Remote host to fetch data: 172.17.0.1
[rbfeeder] [2022-12-15 15:03:17]                Remote port: 30005
[rbfeeder] [2022-12-15 15:03:17]                Remote protocol: BEAST
[rbfeeder] [2022-12-15 15:03:17]  Using GNSS (when available)
[rbfeeder] [2022-12-15 15:03:17]  Start date/time: 2022-12-15 15:03:17
[rbfeeder] [2022-12-15 15:03:17]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:   302 Segmentation fault      qemu-arm-static /usr/bin/rbfeeder_armhf "$@"

mikenye, 2022-12-16:
@doctaste can you post your full container log please?

mikenye, 2022-12-16:
@yitzyb You're definitely running the latest version, which is working for other people... I'm not sure why you're having such a hard time. You could try adding the environment variable `DEBUG_LEVEL=8` to your compose file, restart the container and see if this sheds any light on the problem. If you do this and post your logs, please be sure to sanitise as it will contain sensitive information (your sharing key).

mikenye, 2022-12-16:
@doctaste regarding the armhf - you are indeed pulling the correct image.

Part of all the problems recently was due to the following:

> In a completely baffling turn of events, [Radarbox have decided to drop support for rbfeeder binaries built from their own source repo](https://github.com/airnavsystems/rbfeeder/issues/4#issuecomment-1351342826).
> 
> This PR reverts to the single-architecture (armhf) pre-built rbfeeder binary from Radarbox.
> 
> For non armhf platforms, qemu will be used to execute the binary.
(from #135)

Do you have virtualisation enabled in your system's BIOS? I ask because it seems to segfault when launching qemu...

doctaste, 2022-12-16:
> @doctaste regarding the armhf - you are indeed pulling the correct image.

> Do you have virtualisation enabled in your system's BIOS? I ask because it seems to segfault when launching qemu...

My docker is running on a 64-bit Debian GNU/Linux 11, which runs in an ESXi VM.In BIOS Intel VMX is enabled.

Also according to esxcfg-info VT is enabled.

lscpu doesn't show vmx:

***************************************************************************
root@somehost:~#  lscpu
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   45 bits physical, 48 bits virtual
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              1
Socket(s):                       4
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           142
Model name:                      Intel(R) Core(TM) i5-8279U CPU @ 2.40GHz
Stepping:                        10
CPU MHz:                         2400.000
BogoMIPS:                        4800.00
Hypervisor vendor:               VMware
Virtualization type:             full
L1d cache:                       128 KiB
L1i cache:                       128 KiB
L2 cache:                        1 MiB
L3 cache:                        24 MiB
NUMA node0 CPU(s):               0-3
Vulnerability Itlb multihit:     KVM: Mitigation: VMX unsupported
Vulnerability L1tf:              Mitigation; PTE Inversion
Vulnerability Mds:               Vulnerable: Clear CPU buffers attempted, no microcode; SMT Host state unknown
Vulnerability Meltdown:          Mitigation; PTI
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Retpolines, IBPB conditional, IBRS_FW, STIBP disabled, RSB filling
Vulnerability Srbds:             Unknown: Dependent on hypervisor status
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon nopl xtopology tsc_reliable nonstop_tsc cpuid pni pclmulqdq ssse3 f
                                 ma cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single pti ssbd ibrs ibpb stibp fsgsbase tsc_adjust bmi1 avx2 smep bmi2 in
                                 vpcid rdseed adx smap clflushopt xsaveopt xsavec xgetbv1 xsaves arat flush_l1d arch_capabilities
root@somehost:~#

*********************************************************
But then i realized that i didn't enable the virtualization support in the VM setting itself.

[image]

So enabled it and lscpu now shows a new line "Virtualization" and new CPU flag "vmx":

************************************************************************
Virtualization:                  VT-x

Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon nopl xtopology tsc_reliable nonstop_tsc cpuid pni pclmulqdq vmx sss
                                 e3 fma cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi ept vpid ept_ad fsgsba
                                 se tsc_adjust bmi1 avx2 smep bmi2 invpcid rdseed adx smap clflushopt xsaveopt xsavec xgetbv1 xsaves arat flush_l1d arch_capabilities
*****************************************************
So yes, there was a missing step and now it should be fine.

But the rbfeeder still shows same segfaults:

***********************************************************

[... 45 lines cut ...]
[services.d] done.
[rbfeeder] [2022-12-16 12:12:26]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-16 12:12:26]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-16 12:12:26]  Network-mode enabled.
[rbfeeder] [2022-12-16 12:12:26]                Remote host to fetch data: 172.17.0.1
[rbfeeder] [2022-12-16 12:12:26]                Remote port: 30005
[rbfeeder] [2022-12-16 12:12:26]                Remote protocol: BEAST
[rbfeeder] [2022-12-16 12:12:26]  Using GNSS (when available)
[rbfeeder] [2022-12-16 12:12:26]  Start date/time: 2022-12-16 12:12:26
[rbfeeder] [2022-12-16 12:12:26]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:   301 Segmentation fault      qemu-arm-static /usr/bin/rbfeeder_armhf "$@"
[rbfeeder] [2022-12-16 12:12:27]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-16 12:12:27]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-16 12:12:27]  Network-mode enabled.
[rbfeeder] [2022-12-16 12:12:27]                Remote host to fetch data: 172.17.0.1
[rbfeeder] [2022-12-16 12:12:27]                Remote port: 30005
[rbfeeder] [2022-12-16 12:12:27]                Remote protocol: BEAST
[... cut ...]
[rbfeeder] [2022-12-16 12:12:27]  Start date/time: 2022-12-16 12:12:27
[rbfeeder] [2022-12-16 12:12:27]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:   319 Segmentation fault      qemu-arm-static /usr/bin/rbfeeder_armhf "$@"
[rbfeeder] [2022-12-16 12:12:28]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
**************************************************************

mikenye, 2022-12-17:
@doctaste can you please try running the command `docker run --rm --privileged multiarch/qemu-user-static --reset -p yes -c yes`, and then re-starting the radarbox container?

doctaste, 2022-12-17:
> @doctaste can you please try running the command `docker run --rm --privileged multiarch/qemu-user-static --reset -p yes -c yes`, and then re-starting the radarbox container?

@mikenye 

Yes, sure.

Please find shell output below:

******************************************************************************************

root@somehost# docker compose down    
[+] Running 2/2
 ⠿ Container rbfeeder        Removed                                                                                                                                                                                                                       3.5s
 ⠿ Network radarbox_default  Removed                                                                                                                                                                                                                       0.1s

root@somehost# docker image ls                                                                  
REPOSITORY   TAG       IMAGE ID   CREATED   SIZE

root@somehost# docker run --rm --privileged multiarch/qemu-user-static --reset -p yes -c yes
Unable to find image 'multiarch/qemu-user-static:latest' locally
latest: Pulling from multiarch/qemu-user-static
f5b7ce95afea: Pull complete 
22ee763f0af7: Pull complete 
30abb83a18eb: Pull complete 
0657daef200b: Pull complete 
998118dcc3de: Pull complete 
Digest: sha256:16abbe859db6a7d142d90125f3260008f698a224ccd0e776d554cfc46fd025b6
Status: Downloaded newer image for multiarch/qemu-user-static:latest
Setting /usr/bin/qemu-alpha-static as binfmt interpreter for alpha
Setting /usr/bin/qemu-arm-static as binfmt interpreter for arm
Setting /usr/bin/qemu-armeb-static as binfmt interpreter for armeb
Setting /usr/bin/qemu-sparc-static as binfmt interpreter for sparc
Setting /usr/bin/qemu-sparc32plus-static as binfmt interpreter for sparc32plus
Setting /usr/bin/qemu-sparc64-static as binfmt interpreter for sparc64
Setting /usr/bin/qemu-ppc-static as binfmt interpreter for ppc
Setting /usr/bin/qemu-ppc64-static as binfmt interpreter for ppc64
Setting /usr/bin/qemu-ppc64le-static as binfmt interpreter for ppc64le
Setting /usr/bin/qemu-m68k-static as binfmt interpreter for m68k
Setting /usr/bin/qemu-mips-static as binfmt interpreter for mips
Setting /usr/bin/qemu-mipsel-static as binfmt interpreter for mipsel
Setting /usr/bin/qemu-mipsn32-static as binfmt interpreter for mipsn32
Setting /usr/bin/qemu-mipsn32el-static as binfmt interpreter for mipsn32el
Setting /usr/bin/qemu-mips64-static as binfmt interpreter for mips64
Setting /usr/bin/qemu-mips64el-static as binfmt interpreter for mips64el
Setting /usr/bin/qemu-sh4-static as binfmt interpreter for sh4
Setting /usr/bin/qemu-sh4eb-static as binfmt interpreter for sh4eb
Setting /usr/bin/qemu-s390x-static as binfmt interpreter for s390x
Setting /usr/bin/qemu-aarch64-static as binfmt interpreter for aarch64
Setting /usr/bin/qemu-aarch64_be-static as binfmt interpreter for aarch64_be
Setting /usr/bin/qemu-hppa-static as binfmt interpreter for hppa
Setting /usr/bin/qemu-riscv32-static as binfmt interpreter for riscv32
Setting /usr/bin/qemu-riscv64-static as binfmt interpreter for riscv64
Setting /usr/bin/qemu-xtensa-static as binfmt interpreter for xtensa
Setting /usr/bin/qemu-xtensaeb-static as binfmt interpreter for xtensaeb
Setting /usr/bin/qemu-microblaze-static as binfmt interpreter for microblaze
Setting /usr/bin/qemu-microblazeel-static as binfmt interpreter for microblazeel
Setting /usr/bin/qemu-or1k-static as binfmt interpreter for or1k
Setting /usr/bin/qemu-hexagon-static as binfmt interpreter for hexagon
root@somehost# cat docker-compose.yml
#version: '2.0'

services:
  rbfeeder:
    image: ghcr.io/sdr-enthusiasts/docker-radarbox:latest
    #image: ghcr.io/sdr-enthusiasts/docker-radarbox:v1.0.7-20220512134600_nohealthcheck
    #image: ghcr.io/sdr-enthusiasts/docker-radarbox:v1.0.7-20220528185100_nohealthcheck
    #image: ghcr.io/sdr-enthusiasts/docker-radarbox:v1.0.7-20220531031500_nohealthcheck
    #image: ghcr.io/sdr-enthusiasts/docker-radarbox:v1.0.7-20220719091400
    #image: ghcr.io/sdr-enthusiasts/docker-radarbox:v1.0.7-20220905220100_nohealthcheck
    #image: ghcr.io/sdr-enthusiasts/docker-radarbox:v1.0.8-20220708190411_nohealthcheck_amd64
[... 105 lines cut ...]
[rbfeeder] [2022-12-17 09:05:50]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-17 09:05:50]  Network-mode enabled.
[rbfeeder] [2022-12-17 09:05:50]                Remote host to fetch data: 172.17.0.1
[rbfeeder] [2022-12-17 09:05:50]                Remote port: 30005
[rbfeeder] [2022-12-17 09:05:50]                Remote protocol: BEAST
[rbfeeder] [2022-12-17 09:05:50]  Using GNSS (when available)
[rbfeeder] [2022-12-17 09:05:50]  Start date/time: 2022-12-17 09:05:50
[rbfeeder] [2022-12-17 09:05:50]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:   334 Segmentation fault      /usr/bin/rbfeeder_armhf "$@"
[rbfeeder] [2022-12-17 09:05:51]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-17 09:05:51]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-17 09:05:51]  Network-mode enabled.
[rbfeeder] [2022-12-17 09:05:51]                Remote host to fetch data: 172.17.0.1
[rbfeeder] [2022-12-17 09:05:51]                Remote port: 30005
[... cut ...]
[rbfeeder] [2022-12-17 09:05:51]  Using GNSS (when available)
[rbfeeder] [2022-12-17 09:05:51]  Start date/time: 2022-12-17 09:05:51
[rbfeeder] [2022-12-17 09:05:51]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:   351 Segmentation fault      /usr/bin/rbfeeder_armhf "$@"
root@somehost#
*****************************************************************************************

So still doesn't work.

yitzyb, 2022-12-18:
> @yitzyb You're definitely running the latest version, which is working for other people... I'm not sure why you're having such a hard time. You could try adding the environment variable [...]

I haven't had a chance to make the debug level change, and noticed it connected at some point in the last 2 days.
let's hope it's now fixed.

magnum61, 2023-01-20:
Today i made a new attempt. 
Installed docker from scratch, latest version and started the package with the appropriate command. After five minutes the logs tell me that no data are transferred.

I identified a typo in the command line where i entered a wrong IP address to my feeder device. 
So i stopped the package, fixed the typo and launched the package again.

After that change the log says "Error authenticating Sharing-Key: Invalid sharing-key

As i did not change anyting in the command line beside the IP i don't know how this all comes together now. 
It is probably not a problem of this package but maybe more an error on Radarbox side.

To me the whole service doesn't seem to be reliable

beercity, 2023-01-20:
I recently had that problem. Seems that somehow my sharing key got invalidated. I went to this link and entered my key and now my stats are showing again and the logs are happy
https://www.radarbox.com/sharing-data/claim

magnum61, 2023-01-20:
> I recently had that problem. Seems that somehow my sharing key got invalidated. I went to this link and entered my key and now my stats are showing again and the logs are happy https://www.radarbox.com/sharing-data/claim

WTF?

That was so stupid simple. Why didn't i come to that solution!

Thanks!

closed 2023-02-09 by mikenye