sdr-enthusiasts/docker-airnavradar #130: CPU Serial empty. Could not start connection. Timeout.

qtang, 2022-12-08:
**Issue:** "CPU Serial empty." and "Could not start connection. Timeout." errors using RBFeeder Version 1.0.7 (build 20221027145800) via ghcr.io/sdr-enthusiasts/docker-radarbox:latest and mikenye/radarbox:latest images.

The log entries refer to https://github.com/mikenye/docker-radarbox/blob/master/README.md#workaround-for-cpu-serial, but there is no workaround on that page. However, I have copied cpuinfo into fake_cpuinfo, generated a serial number, added the serial number to the end of fake_cpuinfo, and mapped the fake_cpuinfo volume in the rbfeeder container. I am able to see my serial number when running "cat /proc/cpuinfo" from inside the container.

**Workaround:** Reverting to RBFeeder Version 0.4.3 (build 20210909231001) (image: mikenye/radarbox:0.4.3-20210909231001) fixes issue.

docker logs -f  rbfeeder
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing...
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-rbfeeder: executing...
[cont-init.d] 01-rbfeeder: exited 0.
[cont-init.d] done.
[services.d] starting services
[978json-splitter] 2022/12/08 08:24:11 socat[266] W ioctl(5, IOCTL_VM_SOCKETS_GET_LOCAL_CID, ...): Inappropriate ioctl for device
[978json-splitter] 2022/12/08 08:24:11 socat[266] N listening on AF=2 0.0.0.0:30979
[services.d] done.
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  Starting RBFeeder Version 1.0.7 (build 20221027145800)
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  Network-mode enabled.
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  		Remote host to fetch data: 172.18.0.9
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  		Remote port: 30005
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  		Remote protocol: BEAST
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  Using GNSS (when available)
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  Start date/time: 2022-12-08 08:24:11
[rbfeeder] 2022/12/08 08:24:11 [2022-12-08 08:24:11]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] 2022/12/08 08:24:13 [2022-12-08 08:24:13]  Connection established.
[rbfeeder] 2022/12/08 08:24:13 [2022-12-08 08:24:13]  CPU Serial empty.
[rbfeeder] 2022/12/08 08:24:13 [2022-12-08 08:24:13]  Could not start connection. Timeout.
[978json-splitter] 2022/12/08 08:24:14 socat[266] N accepting connection from AF=2 127.0.0.1:48976 on AF=2 127.0.0.1:30979
[978json-splitter] 2022/12/08 08:24:14 socat[266] N forked off child process 296
[978json-splitter] 2022/12/08 08:24:14 socat[266] N listening on AF=2 0.0.0.0:30979
[978json-splitter] 2022/12/08 08:24:14 socat[296] N opening connection to AF=2 172.18.0.7:30979
[978json-splitter] 2022/12/08 08:24:14 socat[296] N successfully connected from local address AF=2 172.18.0.13:53614
[978json-splitter] 2022/12/08 08:24:14 socat[296] N starting data transfer loop with FDs [6,6] and [5,5]

************ ATTENTION! ************

In order to fix the 'CPU Serial empty' issue you are experiencing,
please see: https://github.com/mikenye/docker-radarbox/blob/master/README.md#workaround-for-cpu-serial

************************************

[rbfeeder] 2022/12/08 08:25:11 [2022-12-08 08:25:11]  ******** Statistics updated every 60 seconds ********
[rbfeeder] 2022/12/08 08:25:11 [2022-12-08 08:25:11]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 0
[rbfeeder] 2022/12/08 08:25:13 [2022-12-08 08:25:13]  Connection established.
[rbfeeder] 2022/12/08 08:25:13 [2022-12-08 08:25:13]  CPU Serial empty.
[rbfeeder] 2022/12/08 08:25:13 [2022-12-08 08:25:13]  Could not start connection. Timeout.
```

Relevant part of compose file:
```
### RadarBox ###
  rbfeeder:
    image: ghcr.io/sdr-enthusiasts/docker-radarbox:latest
#    image: mikenye/radarbox:latest
#    image: mikenye/radarbox:0.4.3-20210909231001
    tty: true
    container_name: rbfeeder
    restart: always
    depends_on:
      - readsb
    environment:
      - BEASTHOST=readsb
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - ALT=${FEEDER_ALT_M}
      - TZ=${FEEDER_TZ}
      - SHARING_KEY=${RADARBOX_SHARING_KEY}
      - UAT_RECEIVER_HOST=dump978
    volumes:
      - '/home/quoc/adsb/data/fake_cpuinfo:/proc/cpuinfo'
    tmpfs:
      - /run:exec,size=64M
      - /var/log
```

Environment:
- Rock64
- Armbian 22.11 Jammy CLI
- Docker version 20.10.21, build baeda1f
- Docker Compose version v2.12.2
- Other services successfully running on same host: readsb, dump978, adsbx, piaware, fr24, pfclient, adsbhub, opensky, tar1090, autoheal, watchtower, lighttpd.

mikenye, 2022-12-15:
This should be resolved by the latest version of the image that was built earlier today. Could you please pull the latest version of the image and see if this fixes your problem. Thanks.

kapiorr, 2022-12-15:
not work for me :(
`[rbfeeder] [2022-12-15 09:13:10]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-15 09:13:10]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-15 09:13:10]  Network-mode enabled.
[rbfeeder] [2022-12-15 09:13:10]  		Remote host to fetch data: 192.168.64.2
[rbfeeder] [2022-12-15 09:13:10]  		Remote port: 30006
[rbfeeder] [2022-12-15 09:13:10]  		Remote protocol: BEAST
[rbfeeder] [2022-12-15 09:13:10]  Using GNSS (when available)
[rbfeeder] [2022-12-15 09:13:10]  Start date/time: 2022-12-15 09:13:10
[rbfeeder] [2022-12-15 09:13:10]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] [2022-12-15 09:13:11]  Connection established.
[rbfeeder] [2022-12-15 09:13:11]  CPU Serial empty. Use MAC address instead.
[rbfeeder] [2022-12-15 09:13:12]  Error authenticating Sharing-Key: Invalid sharing-key
[rbfeeder] [2022-12-15 09:13:21]  Could not start connection. Timeout.
[rbfeeder] [2022-12-15 09:14:10]  ******** Statistics updated every 60 seconds ********
[rbfeeder] [2022-12-15 09:14:10]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 1
[rbfeeder] [2022-12-15 09:14:10]  Data sent: 63 B
[rbfeeder] [2022-12-15 09:14:10]  Data received: 28 B
[rbfeeder] [2022-12-15 09:14:22]  Connection established.
[rbfeeder] [2022-12-15 09:14:22]  CPU Serial empty. Use MAC address instead.
[rbfeeder] [2022-12-15 09:14:22]  Error authenticating Sharing-Key: Invalid sharing-key
[rbfeeder] [2022-12-15 09:14:32]  Could not start connection. Timeout.
[rbfeeder] [2022-12-15 09:15:11]  ******** Statistics updated every 60 seconds ********
[rbfeeder] [2022-12-15 09:15:11]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 2
[rbfeeder] [2022-12-15 09:15:11]  Data sent: 126 B
[rbfeeder] [2022-12-15 09:15:11]  Data received: 56 B
[rbfeeder] [2022-12-15 09:15:38]  Can't connect to AirNav Server. Retry in 60 seconds.
[rbfeeder] [2022-12-15 09:16:12]  ******** Statistics updated every 60 seconds ********
[rbfeeder] [2022-12-15 09:16:12]  Packets sent in the last 60 seconds: 0, Total packets sent since startup: 2
[rbfeeder] [2022-12-15 09:16:12]  Data sent: 126 B
[rbfeeder] [2022-12-15 09:16:12]  Data received: 56 B
`

mikenye, 2022-12-15:
@kapiorr did it eventually connect?

kapiorr, 2022-12-15:
Unfortunately, no

qtang, 2022-12-15:
> This should be resolved by the latest version of the image that was built earlier today. Could you please pull the latest version of the image and see if this fixes your problem. Thanks.

@mikenye It's working for me, thank you for your efforts. It's much appreciated.

@kapiorr I had issues with the 978json-splitter child process dying after I pulled the new container and restarted docker compose. That all went away after I completely rebooted the Rock64. Perhaps give that a try?

Here are the relevant log entries:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing...
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-show-rbfeeder-changelog.sh: executing...

rbfeeder (1:1.0.8+deb11) bullseye; urgency=medium

  * Fix altitude wrong value.
  * Update dump1090 to the latest version.
  * Filter calculated weather values.
  * New field: TAS.
  * New field: Track.
  * New field: True Heading.
  * New field: ADS-B Version.
  * New field: ADS-R Version.
  * New field: TIS-B Version.
  * New field: MRAR Wind Speed.
  * New field: MRAR Wind Dir.
  * New field: MRAR Pressure.
  * New field: MRAR Temperature.
  * New field: MRAR Turbulence.
  * New field: MRAR Humidity.
  * New field: Roll.
  * New command line option: --debug-level (override configuration file).
  * New command line option: --debug-filter <text> to filter debug messages.

 --  Jonis Maurin Ceara <jceara@airnavsystems.com>  Fri, 08 Jul 2022 16:52:56 +0300

[cont-init.d] 01-show-rbfeeder-changelog.sh: exited 0.
[cont-init.d] 02-rbfeeder.sh: executing...
[cont-init.d] 02-rbfeeder.sh: exited 0.
[cont-init.d] 03-show-architecture.sh: executing...

Hardware information:
Machine:   aarch64
Processor: unknown
Platform:  unknown

[cont-init.d] 03-show-architecture.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[978json-splitter] 2022/12/15 09:21:04 socat[291] W ioctl(5, IOCTL_VM_SOCKETS_GET_LOCAL_CID, ...): Inappropriate ioctl for device
[978json-splitter] 2022/12/15 09:21:04 socat[291] N listening on AF=2 0.0.0.0:30979
[rbfeeder] [2022-12-15 09:21:09]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2022-12-15 09:21:09]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2022-12-15 09:21:09]  Network-mode enabled.
[rbfeeder] [2022-12-15 09:21:09]  		Remote host to fetch data: 172.18.0.13
[rbfeeder] [2022-12-15 09:21:09]  		Remote port: 30005
[rbfeeder] [2022-12-15 09:21:09]  		Remote protocol: BEAST
[rbfeeder] [2022-12-15 09:21:09]  Using GNSS (when available)
[rbfeeder] [2022-12-15 09:21:09]  Start date/time: 2022-12-15 09:21:09
[rbfeeder] [2022-12-15 09:21:09]  Socket for ANRB created. Waiting for connections on port 32088
[978json-splitter] 2022/12/15 09:21:12 socat[291] N accepting connection from AF=2 127.0.0.1:41406 on AF=2 127.0.0.1:30979
[978json-splitter] 2022/12/15 09:21:12 socat[291] N forked off child process 313
[978json-splitter] 2022/12/15 09:21:12 socat[291] N listening on AF=2 0.0.0.0:30979
[978json-splitter] 2022/12/15 09:21:12 socat[313] N opening connection to AF=2 172.18.0.11:30979
[978json-splitter] 2022/12/15 09:21:12 socat[313] N successfully connected from local address AF=2 172.18.0.8:52458
[978json-splitter] 2022/12/15 09:21:12 socat[313] N starting data transfer loop with FDs [6,6] and [5,5]
[rbfeeder] [2022-12-15 09:21:15]  Can't connect to AirNav Server. Retry in 60 seconds.
[rbfeeder] [2022-12-15 09:22:15]  Connection established.
[rbfeeder] [2022-12-15 09:22:15]  CPU Serial empty. Use MAC address instead.
[rbfeeder] [2022-12-15 09:22:15]  Error authenticating Sharing-Key: Invalid sharing-key
[rbfeeder] [2022-12-15 09:22:25]  Could not start connection. Timeout.
[rbfeeder] [2022-12-15 09:23:31]  Can't connect to AirNav Server. Retry in 60 seconds.
[rbfeeder] [2022-12-15 09:24:32]  Connection established.
[rbfeeder] [2022-12-15 09:24:32]  CPU Serial empty. Use MAC address instead.
[rbfeeder] [2022-12-15 09:24:33]  Client type: Raspberry Pi
[rbfeeder] [2022-12-15 09:24:33]  Connection with RadarBox24 server OK! Key accepted by server.
[rbfeeder] [2022-12-15 09:24:33]  This is your station serial number: EXTRPI00XXXX //censored by me
[rbfeeder] Thu Dec 15 09:24:39 2022 mlat-client 0.2.12 starting up
[rbfeeder] Thu Dec 15 09:24:39 2022 Listening for Beast-format results connection on port 30105
[rbfeeder] Thu Dec 15 09:24:39 2022 Connected to multilateration server at mlat1.rb24.com:40900, handshaking
[rbfeeder] Thu Dec 15 09:24:40 2022 Beast-format results connection with 127.0.0.1:32004: connection established
[rbfeeder] Thu Dec 15 09:24:42 2022 Server says:
[rbfeeder]
[rbfeeder]         AirNav Server
[rbfeeder]
[rbfeeder]         The multilateration server source code is available under
[rbfeeder]         the terms of the Affero GPL (v3 or later). You may obtain
[rbfeeder]         a copy of this server's source code at the following
[rbfeeder]         location: https://github.com/adsbexchange/mlat-server
[rbfeeder]
[rbfeeder] Thu Dec 15 09:24:42 2022 Handshake complete.
[rbfeeder] Thu Dec 15 09:24:42 2022   Compression:       zlib2
[rbfeeder] Thu Dec 15 09:24:42 2022   UDP transport:     disabled
[rbfeeder] Thu Dec 15 09:24:42 2022   Split sync:        disabled
[rbfeeder] Thu Dec 15 09:24:42 2022 Input connected to 127.0.0.1:32457
[rbfeeder] Thu Dec 15 09:24:42 2022 Input format changed to BEAST, 12MHz clock
[rbfeeder-stats] 2022/12/15 09:26:03 Packets sent in past 5 minutes: 159, total packets since container start: 159
[rbfeeder-stats] 2022/12/15 09:31:05 Packets sent in past 5 minutes: 1349, total packets since container start: 1349
```

mikenye, 2022-12-16:
Thanks @qtang. @kapiorr, can you post the full container logs, censoring your station serial number and any other sensitive data? Thanks.

kapiorr, 2022-12-19:
Today I run container and now works :D
I didn't change anything. It was off for 3 days.

closed 2023-02-05 by mikenye