sdr-enthusiasts/docker-airnavradar #144: Segmentation fault - obviously wrong architecture pulled

magnum61, 2023-01-12:
Hi all,

i am getting this error after pulling the container:

```
[rbfeeder] [2023-01-12 07:52:14]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2023-01-12 07:52:14]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2023-01-12 07:52:14]  Network-mode enabled.
[rbfeeder] [2023-01-12 07:52:14]  		Remote host to fetch data: xxx.xxx.xxx.xxx
[rbfeeder] [2023-01-12 07:52:14]  		Remote port: xxxxx
[rbfeeder] [2023-01-12 07:52:14]  		Remote protocol: BEAST
[rbfeeder] [2023-01-12 07:52:14]  Using GNSS (when available)
[rbfeeder] [2023-01-12 07:52:14]  Start date/time: 2023-01-12 07:52:14
[rbfeeder] [2023-01-12 07:52:14]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[rbfeeder] /usr/bin/rbfeeder: line 17:  4528 Segmentation fault      (core dumped) qemu-arm-static /usr/bin/rbfeeder_armhf "$@"
```

Seem to be that the architecture is identified incorrectly, as it states rbfeeder_armhf, but i am currently running a VM on Proxmox where docker, docker-compose and portainer is installed.
Host is an Intel x86_64 device

How do i get this fixed?
Many thanks in advance

ebfio, 2023-01-12:
Same issue here

ebfio, 2023-01-12:
@magnum61 found the solution: https://github.com/sdr-enthusiasts/docker-radarbox/blob/main/version_0.4.3_workarounds.md

mikenye, 2023-01-12:
Specifically this bit: https://github.com/sdr-enthusiasts/docker-radarbox/blob/main/version_0.4.3_workarounds.md#workaround-for-temperature-sensor

mikenye, 2023-01-13:
I should also note that this isn't an architecture issue - it is poor coding by Radarbox devs.

As Radarbox only make a binary available for armhf architectures, anybody that wants to run the feeder outside of this architecture is forced to run using qemu or another method to run arm binaries on non-arm systems. That is what this container is doing.

When starting the rbfeeder service, it first checks to see if the binary can run natively. If that fails, it attempts to run with qemu.

Within the radarbox binary, rbfeeder attemps to open the file `/sys/class/thermal/thermal_zone0/temp`, without performing any checks or doing any error handling.

On virtual platforms, this file doesn't exist, as the operating system has no access to thermal sensors.

As this file doesn't exist, and as rbfeeder tries to open it anyway - it segfaults.

The workaround linked in my comment above "spoofs" this sensor by making a file that always reads 24 degrees C.

Radarbox should really fix their feeder code.

magnum61, 2023-01-13:
Thanks Mike, thanks ebfio

I agree,  Radarbox didn't design that pretty well.  I am not a coder but followed the discussion since months about their feeder. Looks like they are not really interested in getting issues resolved.

I do have a Jetvision Airsquitter (successor of Radarcape) and installing it directly on the box is also a mess. 

However. I will give it a try later today once my other projects are ongoing and keep you updated here.

magnum61, 2023-01-13:
Yes, it is working!

[image]

Thank You for your support

closed 2023-01-13 by magnum61

mikenye, 2023-01-14:
You’re most welcome!