sdr-enthusiasts/docker-airnavradar #161: Corrupt ini on x86_64 using podman

warthog9, 2023-03-07:
Ran into issues obtaining the sharing key, and saw #77 and #82 and was able to obtain a sharing key via the suggested work around there.

Upon getting the key I continued following the instructions from https://sdr-enthusiasts.gitbook.io/ads-b/feeder-containers/feeding-radarbox and completed my docker-compose.yaml with the appropriate section.

relevant section of docker compose:
```
  rbfeeder:
    labels:
      io.containers.autoupdate: registry
    image: ghcr.io/sdr-enthusiasts/docker-radarbox:latest
    tty: true
    container_name: rbfeeder
    restart: always
    depends_on:
      - readsbu
    environment:
      - BEASTHOST=readsb
      - LAT=${FEEDER_LAT}
      - LONG=${FEEDER_LONG}
      - ALT=${FEEDER_ALT_M}
      - TZ=${FEEDER_TZ}
      - SHARING_KEY=${RADARBOX_SHARING_KEY}
    tmpfs:
      - /run:exec,size=64M
      - /var/log
```

Upon startup of the container however it basically runs into nothing bot log errors:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing... 
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-show-rbfeeder-changelog.sh: executing... 

rbfeeder (1:1.0.8+deb11) bullseye; urgency=medium

  * Fix altitude wrong value.
  * Update dump1090 to the latest version.
  * Filter calculated weather values. 
  * New field: TAS.
  * New field: Track.
  * New field: True Heading.
  * New field: ADS-B Version.
  * New field: ADS-R Version.
  * New field: TIS-B Version.
  * New field: MRAR Wind Speed.
  * New field: MRAR Wind Dir.
  * New field: MRAR Pressure.
  * New field: MRAR Temperature.
  * New field: MRAR Turbulence.
  * New field: MRAR Humidity.
  * New field: Roll.
  * New command line option: --debug-level (override configuration file).
  * New command line option: --debug-filter <text> to filter debug messages.

 --  Jonis Maurin Ceara <jceara@airnavsystems.com>  Fri, 08 Jul 2022 16:52:56 +0300

[cont-init.d] 01-show-rbfeeder-changelog.sh: exited 0.
[cont-init.d] 02-rbfeeder.sh: executing... 
[cont-init.d] 02-rbfeeder.sh: exited 0.
[cont-init.d] 03-show-architecture.sh: executing... 

Hardware information:
Machine:   x86_64
Processor: unknown
Platform:  unknown

[cont-init.d] 03-show-architecture.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Starting RBFeeder Version 1.0.8 (build 20220708190411)
[rbfeeder] [2023-03-06 21:33:28]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2023-03-06 21:33:28]  Network-mode disabled. Using local dongle.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Using GNSS (when available)
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file (/etc/rbfeeder.ini).
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Start date/time: 2023-03-06 21:33:28
[rbfeeder] [2023-03-06 21:33:28]  Socket for ANRB created. Waiting for connections on port 32088
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] [2023-03-06 21:33:28]  Error loading ini file.
[rbfeeder] /usr/bin/dump1090-rb: error while loading shared libraries: libncurses.so.6: cannot open shared object file: No such file or directory
```

This looks like it's failing in two directions to me (1) that there's something wrong with the ini file, though it's clear why, the

kx1t, 2023-03-07:
it looks like somehow the `external_host` IP address appeared in it twice:
```
external_host=10.89.0.32
10.89.0.32
```

If this happens again after a container recreation, could you check what the following does?
```
docker exec -it rbfeeder bash -c '/bin/s6-dnsip4 $BEASTHOST'
```
It should return a clean, single, IP address (probably `10.89.0.32` in your case).

Note - I checked the code and there's no reason why it should print that IP address twice into your config file 🤷‍♂️

warthog9, 2023-03-07:
```
# podman exec -it rbfeeder bash -c '/bin/s6-dnsip4 $BEASTHOST'
10.89.0.32
10.89.0.32
```

Apparently somewhere in the process aardvark-dns has gotten two IPs in it for the same host (not entirely sure how, but it's technically not invalid since that's allowed for DNS)

```
# dig @10.89.0.1 readsb

; <<>> DiG 9.18.12 <<>> @10.89.0.1 readsb
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 27167
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
; COOKIE: d6beaa306d6003a8 (echoed)
;; QUESTION SECTION:
;readsb.                                IN      A

;; ANSWER SECTION:
readsb.                 86400   IN      A       10.89.0.32
readsb.                 86400   IN      A       10.89.0.32

;; Query time: 0 msec
;; SERVER: 10.89.0.1#53(10.89.0.1) (UDP)
;; WHEN: Tue Mar 07 09:22:50 PST 2023
;; MSG SIZE  rcvd: 79
```

dirkhh, 2023-03-07:
I read elsewhere that there's a problem with `rbfeeder` crashing when you don't have a temperature sensor... 
https://discord.com/channels/734090820684349521/734090821250580564/1062399819706019920
But I think this one is a different issue.

mikenye, 2023-03-24:
@warthog9 if you pull the latest image and re-create your container, this should be fixed.

closed 2023-03-24 by mikenye

warthog9, 2023-03-27:
confirmed fixed!

mikenye, 2023-03-27:
Thanks mate!