sdr-enthusiasts/docker-airnavradar #170: Could not resolve host name specified in 'external_host'.

Freekers, 2023-03-24:
The ```latest``` image got updated last night and since then my rbfeeder is not working anymore. The logs are full with the following error messages:
```
[rbfeeder] getaddrinfo: Name or service not known
[rbfeeder] [2023-03-24 10:40:59]  Could not resolve host name specified in 'external_host'.
[rbfeeder] getaddrinfo: Name or service not known
[rbfeeder] [2023-03-24 10:41:00]  Could not resolve host name specified in 'external_host'.
[rbfeeder] getaddrinfo: Name or service not known
```
This used to work fine before the last update. I think this commit might be related: https://github.com/sdr-enthusiasts/docker-radarbox/commit/965f76b329b938cc21eaeb07953790f32b74875f

For reference, here is my docker compose:

```
  rbfeeder:
    image: ghcr.io/sdr-enthusiasts/docker-radarbox:latest
    container_name: rbfeeder
    restart: unless-stopped
    environment:
      - TZ=Europe/Amsterdam
      - BEASTHOST=192.168.30.224
      - LAT=<REDACTED>
      - LONG=<REDACTED>
      - ALT=25
      - SHARING_KEY=<REDACTED>
      - ENABLE_MLAT=false
   ```

Thanks

mikenye, 2023-03-24:
Oh I'm really sorry if I've broken it for you. 

Can you please run `docker exec -it radarbox cat /etc/rbfeeder.ini` please?

Censor your feeder key and location and post here. 

Thanks.

Freekers, 2023-03-24:
No worries, here's the output of the command

```
cat /etc/rbfeeder.ini
[client]
network_mode=true
log_file=/var/log/rbfeeder.log
debug_level=0

key=<REDACTED>
lat=<REDACTED>
lon=<REDACTED>
alt=25

[network]
mode=beast
  
external_port=30005
external_host=
[mlat]
autostart_mlat=false
```

Seems like the external_host value is missing

kapiorr, 2023-03-24:
same for me :(

CaptainSteubing, 2023-03-24:
I had to shell into the running container and run this command to temporarily fix it.

`sed -i 's/external_host=/external_host=<your_feeder_ip>/g' /etc/rbfeeder.ini `

Substituting <your_feeder_ip> with your ADSB feeder's IP.

mikenye, 2023-03-24:
Once again apologies for the inconvenience.

I've just merged a fix for this. It will take a few minutes to build the new image.

After getting updating to the new image, please let me know if the problem still occurs.

closed 2023-03-24 by mikenye

mikenye, 2023-03-24:
Images are built! `ghcr.io/sdr-enthusiasts/docker-radarbox:latest`.

Freekers, 2023-03-25:
Thanks for the quick fix. I can confirm the issue is now resolved :)

mikenye, 2023-03-26:
Thanks for letting me know!