sdr-enthusiasts/docker-airnavradar #172: unable to create sharing key

yousaf465, 2023-06-25:
while running the docker container on rpi4 

```
[radarbox-feeder] [2023-06-24 23:04:17]  Data received: 629 B
[radarbox-feeder] [2023-06-24 23:04:44]  Connection established.
[radarbox-feeder] [2023-06-24 23:04:44]  Empty sharing key. We will try to create a new one for you!
[radarbox-feeder] [2023-06-24 23:04:54]  Timeout waiting for new key. Will try again in 30 seconds.
[radarbox-feeder] [2023-06-24 23:05:18]  ******** Statistics updated every 60 seconds ********
[radarbox-feeder] [2023-06-24 23:05:18]  Packets sent in the last 60 seconds: 1, Total packets sent since startup: 27
[radarbox-feeder] [2023-06-24 23:05:18]  Data sent: 675 B
[radarbox-feeder] [2023-06-24 23:05:18]  Data received: 629 B
```
how to fix this?

mpsparrow, 2023-06-28:
Having the exact same issue and have yet to find a solution.

mpsparrow, 2023-06-30:
I had a similar issue a few days ago. Tried again today and after a minute I was given a key.

yousaf465, 2023-06-30:
> I had a similar issue a few days ago. Tried again today and after a minute I was given a key.

thanks got the sharing key today

`
    ghcr.io/sdr-enthusiasts/docker-radarbox:latest
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing...
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-show-rbfeeder-changelog.sh: executing...

rbfeeder (1:1.0.8+deb11) bullseye; urgency=medium

  * Fix altitude wrong value.
  * Update dump1090 to the latest version.
  * Filter calculated weather values.
  * New field: TAS.
  * New field: Track.
  * New field: True Heading.
  * New field: ADS-B Version.
  * New field: ADS-R Version.
  * New field: TIS-B Version.
  * New field: MRAR Wind Speed.
  * New field: MRAR Wind Dir.
  * New field: MRAR Pressure.
  * New field: MRAR Temperature.
  * New field: MRAR Turbulence.
  * New field: MRAR Humidity.
  * New field: Roll.
  * New command line option: --debug-level (override configuration file).
  * New command line option: --debug-filter <text> to filter debug messages.

 --  Jonis Maurin Ceara <jceara@airnavsystems.com>  Fri, 08 Jul 2022 16:52:56 +0                                                                                                300

[cont-init.d] 01-show-rbfeeder-changelog.sh: exited 0.
[cont-init.d] 02-rbfeeder.sh: executing...
WARNING: TZ environment variable not set

WARNING: SHARING_KEY environment variable was not set!
Please make sure you note down the key generated.
Pass the key as environment var SHARING_KEY on next launch!

[cont-init.d] 02-rbfeeder.sh: exited 0.
[cont-init.d] 03-show-architecture.sh: executing...

Hardware information:
Machine:   aarch64
Processor: unknown
Platform:  unknown

[cont-init.d] 03-show-architecture.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[rbfeeder] [2023-06-30 17:41:02]  Starting RBFeeder Version 1.0.8 (build 2022070                                                                                                8190411)
[rbfeeder] [2023-06-30 17:41:02]  Using configuration file: /etc/rbfeeder.ini
[rbfeeder] [2023-06-30 17:41:02]  Network-mode enabled.
[rbfeeder] [2023-06-30 17:41:02]                Remote host to fetch data: 127.0                                                                                                .0.1
[rbfeeder] [2023-06-30 17:41:02]                Remote port: 30005
[rbfeeder] [2023-06-30 17:41:02]                Remote protocol: BEAST
[rbfeeder] [2023-06-30 17:41:02]  Using GNSS (when available)
[rbfeeder] [2023-06-30 17:41:02]  Start date/time: 2023-06-30 17:41:02
[rbfeeder] [2023-06-30 17:41:02]  Socket for ANRB created. Waiting for connectio                                                                                                ns on port 32088
[rbfeeder] [2023-06-30 17:41:03]  Connection established.
[rbfeeder] [2023-06-30 17:41:03]  Empty sharing key. We will try to create a new                                                                                                 one for you!
[rbfeeder] [2023-06-30 17:41:04]  Your new key is xxxxxxxxxxxxxxxxxxxxxxxx                                                                                                79. Please save this key for future use. You will have to know this key to link                                                                                                 this receiver to your account in RadarBox24.com. This key is also saved in confi                                                                                                guration file (/etc/rbfeeder.ini)
^C[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
s6-svwait: fatal: timed out
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
`

now I am confused 79 is included in key or not?

kx1t, 2023-11-21:
This issue is probably related to the following problem of the RadarBox server. Unfortunately we can't fix it; our advice is to keep on trying until you get a response. Generally, the DNS cache is 300 seconds, so it will go to a different IP address after about 5 minutes.

Your sharing key should be 32 hex digits long.

> 
> It appears that when connecting to the RB server, the server address (rpiserver-ng.rb24.com) circles through the 6  different IP addresses listed below (from: “dig rpiserver-ng.rb24.com a”). Several of these IP addresses aren’t connected to a working RadarBox server - the client will connect but then rejects the Sharing Key. This is what we found (consistently replicable):
> 
> good: 79.133.49.85
> good: 23.29.123.254
> bad: 23.29.123.240
> bad: 46.21.149.110
> untested: 79.133.49.90
> untested: 46.21.149.109

closed 2023-11-21 by kx1t