sdr-enthusiasts/docker-airnavradar #183: [rbfeeder] ERROR: Could not run rbfeeder natively or via qemu

totalhavoc, 2023-11-20:
I just deployed and I'm getting the following error when running the container. 

[rbfeeder] ERROR: Could not run rbfeeder natively or via qemu

I'm running on bookworm so I'm not sure if that could be the issue or not.

kx1t, 2023-11-21:
Can you share the hardware architecture you are running this on? And the base system’s OS/version? VM or no VM? Whatever the following commands are telling you:

```bash
cat /etc/os-release
uname-a
```

Generally, this container should run in Linux armhf, arm64, or amd64. It will run on some VMs as well but we don’t really test for this.

kx1t, 2023-11-21:
Bookworm shouldn’t be an issue. That’s what the container itself is based on. 
The issue that we have come across is that RadarBox doesn’t provide any usable source code for us to build from and we’re stuck using their pre compiled binaries, with qemu if needed 🤷‍♂️

kx1t, 2023-11-21:
Final update for the evening: we discovered that TODAY AIrNav had brought out a new version of `rbfeeder` for armhf. I reworked the code a bit (see this PR https://github.com/sdr-enthusiasts/docker-radarbox/pull/184), so that:
- on `armhf` platforms, it will use their `armhf` build natively
- on `arm64` platforms (like yours), it will use their `arm64` build natively
- on `amd64` platforms, it will use their `armhf` build in `qemu`

Please follow the PR - once merged in (our process says that one of use, not the author, needs to approve major changes), it will kick off an auto-build that should hopefully fix the issue for you.

I tested the new build on a few platforms incl arm64/Bookworm and it worked for me.

totalhavoc, 2023-11-21:
Thanks for looking into this. Sorry I didn't respond earlier but I had set this aside to work on other projects. I will let you know how this goes as soon as I can this morning

closed 2023-11-21 by totalhavoc