sdr-enthusiasts/docker-airnavradar #193: Error Segmentation fault      (core dumped) qemu-arm-static /usr/bin/rbfeeder_arm "$@"

Taubin, 2024-01-22:
I have set this up previously with no issue, but when I moved to a new vm, I keep getting the following error when running rbfeeder

This isn't an arm machine (it's an old pc running proxmox which ran this prior fine). I tried pulling just the amd version using the amd tag and got the same error. 

```
rbfeeder  | [2024-01-22 16:10:57.722][rbfeeder] [2024-01-22 16:10:57]  Starting RBFeeder Version 1.0.10 (build 20231120150000)
rbfeeder  | [2024-01-22 16:10:57.723][rbfeeder] [2024-01-22 16:10:57]  Using configuration file: /etc/rbfeeder.ini
rbfeeder  | [2024-01-22 16:10:57.723][rbfeeder] [2024-01-22 16:10:57]  Network-mode enabled.
rbfeeder  | [2024-01-22 16:10:57.723][rbfeeder] [2024-01-22 16:10:57]           Remote host to fetch data: 172.18.0.6
rbfeeder  | [2024-01-22 16:10:57.723][rbfeeder] [2024-01-22 16:10:57]           Remote port: 30005
rbfeeder  | [2024-01-22 16:10:57.723][rbfeeder] [2024-01-22 16:10:57]           Remote protocol: BEAST
rbfeeder  | [2024-01-22 16:10:57.734][rbfeeder] [2024-01-22 16:10:57]  Using GNSS (when available)
rbfeeder  | [2024-01-22 16:10:57.741][rbfeeder] [2024-01-22 16:10:57]  Start date/time: 2024-01-22 16:10:57
rbfeeder  | [2024-01-22 16:10:57.761][rbfeeder] [2024-01-22 16:10:57]  Socket for ANRB created. Waiting for connections on port 32088
rbfeeder  | [2024-01-22 16:10:57.771][rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
rbfeeder  | [2024-01-22 16:10:57.876][rbfeeder] /usr/bin/rbfeeder: line 20:  3694 Segmentation fault      (core dumped) qemu-arm-static /usr/bin/rbfeeder_arm "$@"
rbfeeder  | [2024-01-22 16:10:57.876][rbfeeder] [ERROR] /usr/bin/rbfeeder in line 15 when executing: qemu-arm-static /usr/bin/rbfeeder_arm "$@"
```

Docker-compose: 

```
services:
  ultrafeeder:
  # ultrafeeder combines a number of functions:
  # - it retrieves and decodes 1090MHz Mode A/C/S data from the SDR(s) using Wiedehopf's branch of readsb
  # - it implements a `tar1090` based map on port 80 (mapped to port 8080 on the host)
  # - it includes graph1090 (system statistics website) on http://xxxxx/graphs1090
  # - it sends ADSB data directly (without the need of additional containers) to the
  #   "new" aggregators, and, if desired, also to AdsbExchange
  # - it includes mlat-client to send MLAT data to these aggregators
  # - it includes an MLAT Hub to consolidate MLAT results and make them available to the built-in map and other services

    image: ghcr.io/sdr-enthusiasts/docker-adsb-ultrafeeder
    tty: true
    container_name: ultrafeeder
    hostname: ultrafeeder
    restart: unless-stopped
    device_cgroup_rules:
      - 'c 189:* rwm'
    ports:
      - 8080:80               # to expose the web interface
    environment:
      # --------------------------------------------------
      # general parameters:
      - LOGLEVEL=error
      - TZ=${FEEDER_TZ}
      # --------------------------------------------------
      # SDR related parameters:
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_RTLSDR_DEVICE=${ADSB_SDR_SERIAL}
      - READSB_RTLSDR_PPM=${ADSB_SDR_PPM}
      #
      # --------------------------------------------------
      # readsb/decoder parameters:
      - READSB_LAT=${FEEDER_LAT}
      - READSB_LON=${FEEDER_LONG}
      - READSB_ALT=${FEEDER_ALT_M}m
      - READSB_GAIN=${ADSB_SDR_GAIN}
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      #
      # --------------------------------------------------
      # Sources and Aggregator connections:
      # Note - remove the ones you are not using / feeding
      #  Make sure that each line ends with a semicolon ";",  with the
      #  exception of the last line which shouldn't have a ";"
      - ULTRAFEEDER_CONFIG=
          adsb,dump978,30978,uat_in;
          adsb,feed.adsb.fi,30004,beast_reduce_plus_out;
[... 171 lines cut ...]
cache size      : 16384 KB
physical id     : 1
siblings        : 2
core id         : 1
cpu cores       : 2
apicid          : 3
initial apicid  : 3
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm constant_tsc nopl xtopology cpuid tsc_known_freq pni ssse3 cx16 sse4_1 sse4_2 x2apic popcnt aes hypervisor lahf_lm cpuid_fault pti
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips        : 6185.67
clflush size    : 64
cache_alignment : 128
address sizes   : 40 bits physical, 48 bits virtual
power management:

serial          : 1b25acdcf052e6ab
```

I can share my .env if really needed but it will all be redacted for obvious reasons. I've verified the key and naming convention within the env file. 

I'm sure it's something dumb I've done, but I can't figure out what it might be.  Any help would be greatly appreciated. Thank you.

ctschach, 2024-03-05:
I have the exact same problem running it on a virtual machine hosted on Proxmox:
```
[2024-03-05 09:47:22.872][rbfeeder] [2024-03-05 09:47:22]  Starting RBFeeder Version 1.0.10 (build 20231120150000)
[2024-03-05 09:47:22.873][rbfeeder] [2024-03-05 09:47:22]  Using configuration file: /etc/rbfeeder.ini
[2024-03-05 09:47:22.874][rbfeeder] [2024-03-05 09:47:22]  Network-mode enabled.
[2024-03-05 09:47:22.874][rbfeeder] [2024-03-05 09:47:22]  		Remote host to fetch data: 192.168.42.178
[2024-03-05 09:47:22.874][rbfeeder] [2024-03-05 09:47:22]  		Remote port: 30005
[2024-03-05 09:47:22.874][rbfeeder] [2024-03-05 09:47:22]  		Remote protocol: BEAST
[2024-03-05 09:47:22.881][rbfeeder] [2024-03-05 09:47:22]  Using GNSS (when available)
[2024-03-05 09:47:22.887][rbfeeder] [2024-03-05 09:47:22]  Start date/time: 2024-03-05 09:47:22
[2024-03-05 09:47:22.900][rbfeeder] [2024-03-05 09:47:22]  Socket for ANRB created. Waiting for connections on port 32088
[2024-03-05 09:47:22.907][rbfeeder] qemu: uncaught target signal 11 (Segmentation fault) - core dumped
[2024-03-05 09:47:22.966][rbfeeder] /usr/bin/rbfeeder: line 20:  5117 Segmentation fault      (core dumped) qemu-arm-static /usr/bin/rbfeeder_arm "$@"
[2024-03-05 09:47:22.966][rbfeeder] [ERROR] /usr/bin/rbfeeder in line 15 when executing: qemu-arm-static /usr/bin/rbfeeder_arm "$@"
```

Taubin, 2024-03-13:
> Okay, here is the trick - create a folder on the host machine for mounting the necessary temperature.
> <snip> 
> </snip>
> However, the above approach works for me.

Unfortunately, that didn't work for me. I've had it added since the initial installation. I believe it's noted in the install docs as well.

kx1t, 2024-04-09:
I don't think we can currently fix this, closing this issue

closed 2024-04-09 by kx1t